genodelabs/genode #5450: hw: do not "trap" into kernel exception entry just to initialize Vcpu context

skalk, 2025-02-12:
While working at issue #5425 I stumbled across that x86 virtualization code in base-hw,which - without any needs - jumps into the exception vector of this architecture to handle a special case of Vcpu context (VMCB etc.) initialization. After offline discussion with @atopia it became clear that it is not needed to do so. The only fact that remains is there needs to some initialization code executed by the target cpu before the guest code is run the first time. The virtualization code should avoid to do this first pseudo trap.

skalk, 2025-02-12:
@atopia please have a look at commit df14567, I made some minor changes (enum renaming, line spaces,...) to your original commit. I hope you're fine with it.

atopia, 2025-02-17:
@skalk Looks good, thank you!

chelmuth, 2025-02-26:
x86_64.pc.hw.vmm_x86 fails reproducably with the following log.

```
Kernel panic on CPU 1
Exception reason is undefined instruction (trapno=6)

Register dump:

ip     = 0xffffffc00003dfe4
sp     = 0xffffffe07010fc50
cs     = 0x8
ss     = 0x0
eflags = 0x10006
rax    = 0x1001c7000
rbx    = 0xffffffd0010c6000
rcx    = 0xc0000103
rdx    = 0x0
rdi    = 0xffffffd0010c3000
rsi    = 0xffffffd0010c6000
rbp    = 0xffffffd0010c3000
CR2    = 0x92f40
CR3    = 0x153000
```

closed 2025-02-27 by chelmuth