genodelabs/genode #5455: insufficient kernel memory quota

cproc, 2025-02-17:
I'm currently testing the Falkon web browser on current staging Sculpt and often get errors like the following:

```
[ 0] sys_pd_ctrl: PD 0xffffffff810306a0 has insufficient kernel memory quota                                                                                 
Warning: kernel memory quota upgrade failed - trigger memory free up for causing 'init -> runtime -> falkon-jemalloc -> falkon':'pthread.12' - donator is 'core':'unknown', policy=2
```

I'm not sure how to fix this.

alex-ab, 2025-02-26:
Actually it should be just a warning, does Falkon afterwards still work or is it done ?  What does 162cdc4492b8232ef2ef79c932ba956dbc1ae1af show for numbers  regarding the kernel resource usage of Falkon.

cproc, 2025-02-26:
Falkon does not respond to mouse input anymore when it happens and the log messages keep repeating every 2 seconds or so:

```
[ 1] sys_pd_ctrl: PD 0xffffffff81030660 has insufficient kernel memory quota                                                                                 
Warning: kernel memory quota upgrade failed - trigger memory free up for causing 'init -> runtime -> falkon-jemalloc -> falkon':'pthread.13' - donator is 'co
re':'unknown', policy=2
Warning: usage before=34989 after=34988 - limit before=34992 after=34992
[ 1] sys_pd_ctrl: PD 0xffffffff81030660 has insufficient kernel memory quota
Warning: kernel memory quota upgrade failed - trigger memory free up for causing 'init -> runtime -> falkon-jemalloc -> falkon':'pthread.13' - donator is 'co
re':'unknown', policy=2
Warning: usage before=34989 after=34989 - limit before=34992 after=34992
[ 1] sys_pd_ctrl: PD 0xffffffff81030660 has insufficient kernel memory quota
Warning: kernel memory quota upgrade failed - trigger memory free up for causing 'init -> runtime -> falkon-jemalloc -> falkon':'pthread.13' - donator is 'co
re':'unknown', policy=2
Warning: usage before=34989 after=34989 - limit before=34992 after=34992
[ 0] sys_pd_ctrl: PD 0xffffffff81030660 has insufficient kernel memory quota
Warning: kernel memory quota upgrade failed - trigger memory free up for causing 'init -> runtime -> falkon-jemalloc -> falkon':'signal handler' - donator is
 'core':'unknown', policy=2
Warning: usage before=34989 after=34989 - limit before=34992 after=34992
[ 1] sys_pd_ctrl: PD 0xffffffff81030660 has insufficient kernel memory quota
Warning: kernel memory quota upgrade failed - trigger memory free up for causing 'init -> runtime -> falkon-jemalloc -> falkon':'pthread.13' - donator is 'co
re':'unknown', policy=2
Warning: usage before=34989 after=34989 - limit before=34992 after=34992
[ 0] sys_pd_ctrl: PD 0xffffffff81030660 has insufficient kernel memory quota
Warning: kernel memory quota upgrade failed - trigger memory free up for causing 'init -> runtime -> falkon-jemalloc -> falkon':'signal handler' - donator is
 'core':'unknown', policy=2
Warning: usage before=34989 after=34989 - limit before=34992 after=34992
```

alex-ab, 2025-02-28:
@cproc: Thanks, may you please replace the debug commit with 544dc59bc86a8868d5e90ce5eedb4cf51cf6772d and please re-run ?

cproc, 2025-02-28:
This is the output with 544dc59:

```
[ 0] sys_pd_ctrl: PD 0xffffffff81030a20 has insufficient kernel memory quota                                                                                 
[ 0] 0xffffffff81030a20 usage/limit 1184/1441 o=0                                                                                                            
[ 0] 0xffffffff81030a20 pt : slabs=751 elements=15531/15771, suspicious=0                                                                                    
[ 0] 0xffffffff81030a20 mdb: slabs=122 elements=3390/3416, suspicious=0                                                                                      
[ 0] 0xffffffff81030a20 sm : slabs=100 elements=1763/1800, suspicious=0                                                                                      
[ 0] 0xffffffff81030a20 sc : slabs=2 elements=6/36, suspicious=0                                                                                             
[ 0] 0xffffffff81030a20 ec : slabs=4 elements=18/24, suspicious=0                                                                                            
[ 0] 0xffffffff81030a20 fpu: slabs=9 elements=9/9, suspicious=0
[ 0] 0xffffffff81bca540 usage/limit 35159/35161 o=0
[ 0] 0xffffffff81bca540 pt : slabs=1 elements=3/21, suspicious=0
[ 0] 0xffffffff81bca540 mdb: slabs=1084 elements=30326/30352, suspicious=0
[ 0] 0xffffffff81bca540 sm : slabs=2 elements=12/36, suspicious=0
[ 0] 0xffffffff81bca540 sc : slabs=4 elements=53/72, suspicious=0
[ 0] 0xffffffff81bca540 ec : slabs=12 elements=54/72, suspicious=0
[ 0] 0xffffffff81bca540 fpu: slabs=68 elements=53/68, suspicious=0
Warning: kernel memory quota upgrade failed - trigger memory free up for causing 'init -> runtime -> falkon-jemalloc -> falkon':'pthread.12' - donator is 'core':'unknown', policy=2
Warning: usage before=35159 after=35159 - limit before=35161 after=35161
[ 0] sys_pd_ctrl: PD 0xffffffff81030a20 has insufficient kernel memory quota
[ 0] 0xffffffff81030a20 usage/limit 1184/1441 o=0
[ 0] 0xffffffff81030a20 pt : slabs=751 elements=15435/15771, suspicious=0
[ 0] 0xffffffff81030a20 mdb: slabs=122 elements=3041/3416, suspicious=0
[ 0] 0xffffffff81030a20 sm : slabs=100 elements=1763/1800, suspicious=0
[ 0] 0xffffffff81030a20 sc : slabs=2 elements=6/36, suspicious=0
[ 0] 0xffffffff81030a20 ec : slabs=4 elements=18/24, suspicious=0
[ 0] 0xffffffff81030a20 fpu: slabs=9 elements=9/9, suspicious=0
[ 0] 0xffffffff81bca540 usage/limit 35159/35161 o=0
[ 0] 0xffffffff81bca540 pt : slabs=1 elements=3/21, suspicious=0
[ 0] 0xffffffff81bca540 mdb: slabs=1084 elements=30001/30352, suspicious=0
[ 0] 0xffffffff81bca540 sm : slabs=2 elements=3/36, suspicious=0
[ 0] 0xffffffff81bca540 sc : slabs=4 elements=53/72, suspicious=0
[ 0] 0xffffffff81bca540 ec : slabs=12 elements=54/72, suspicious=0
[ 0] 0xffffffff81bca540 fpu: slabs=68 elements=53/68, suspicious=0
Warning: kernel memory quota upgrade failed - trigger memory free up for causing 'init -> runtime -> falkon-jemalloc -> falkon':'signal handler' - donator is
 'core':'unknown', policy=2
Warning: usage before=35159 after=35159 - limit before=35161 after=35161
```

alex-ab, 2025-02-28:
@cproc: can you please try to adjust in repos/base-nova/src/kernel/nova/target.mk the value of CONFIG_MEMORY_DYN_PER_MILL to some higher value (default is 1% of system memory is for the kernel), just to see whether the kernel memory ever will be sufficient. core will report the effective kernel memory during boot.

alex-ab, 2025-03-04:
```
Allocator 0xb45458 dump:
 => mem_size=140736414609408 (134216703 MB) / mem_avail=140734792507392 (134215157 MB), mem_used =1622102016 (1546 MB)
oom - 'init -> runtime -> falkon-jemalloc -> falkon':'pthread.13' (37047/37049) - transfer 2 pages from 'core':'unknown' (1334/164028)
PD (init -> runtime -> falkon-jemalloc -> falkon) released dataspace cap (used=973, limit=9296)
oom - 'init -> runtime -> falkon-jemalloc -> falkon':'pthread.13' (37049/37051) - transfer 2 pages from 'core':'unknown' (1334/164026)
PD (init -> runtime -> falkon-jemalloc -> falkon) released dataspace cap (used=972, limit=9296)
PD (init -> runtime -> falkon-jemalloc -> falkon) released dataspace cap (used=971, limit=9296)
PD (init -> runtime -> falkon-jemalloc -> falkon) released dataspace cap (used=970, limit=9296)
PD (init -> runtime -> falkon-jemalloc -> falkon) released dataspace cap (used=969, limit=9296)
[init -> runtime -> falkon-jemalloc -> falkon] 
[init -> runtime -> falkon-jemalloc -> falkon] <--- Last few GCs --->
[init -> runtime -> falkon-jemalloc -> falkon] 
[init -> runtime -> falkon-jemalloc -> falkon] [1:0x1a13df30]   432703 ms: Mark-sweep (reduce) 252.5 (257.5) -> 252.5 (257.5) MB, 16.5 / 0.1 ms  (average mu = 0.829, current mu = 0.001) last resort GC in old space requested
[init -> runtime -> falkon-jemalloc -> falkon] [1:0x1a13df30]   432721 ms: Mark-sweep (reduce) 252.5 (257.5) -> 252.5 (257.5) MB, 18.1 / 0.1 ms  (average mu = 0.681, current mu = 0.001) last resort GC in old space requested
[init -> runtime -> falkon-jemalloc -> falkon] 
[init -> runtime -> falkon-jemalloc -> falkon] 
[init -> runtime -> falkon-jemalloc -> falkon] <--- JS stacktrace --->
[init -> runtime -> falkon-jemalloc -> falkon] 
[init -> runtime -> falkon-jemalloc -> falkon] [1:436677040:0304/115903.434000:FATAL:memory.cc(38)] Out of memory. size=0
Warning: unresolvable exception 3, pd 'init -> runtime -> falkon-jemalloc -> falkon', thread 'pthread.13', cpu 3, ip=0x4c4910a sp=0x411f81e0 bp=0x411f86d0 no signal handler
PD (init -> runtime -> falkon-jemalloc -> falkon) released dataspace cap (used=968, limit=9296)
PD (init -> runtime -> falkon-jemalloc -> falkon) released dataspace cap (used=967, limit=9296)
```

When increasing the kernel memory to 5 % of system ram Falkon will continue until there is some out of memory in Falkon. Nevertheless, I will try to understand as side project whether the numbers regarding the kernel quota make sense.

cproc, 2025-03-04:
I did some more tests with tne non-jemalloc Falkon variant and a run script without Sculpt last week and adapted the memory limits in QtWebEngine to avoid the JavaScript out of memory error and found so far that the benchmark finished successfully with `-DCONFIG_MEMORY_DYN_MIN=0x17000000` but not with `-DCONFIG_MEMORY_DYN_MIN=0x16000000`. That's about 368M.

alex-ab, 2025-03-11:
I added on my issue_5455 branch more debugging commits to the kernel, which now shows that Falkon uses a wide variety of virtual addresses  (for me more then 129 x 1GB ranges) which explains the large kernel resource usage. Restricting the available virtual address space from close to unlimited to some smaller value, e.g. as done in a762ff07df3168ae04d96a48115a09141565ef90 let Falkon continue until the JS stacktrace issue for me.

cproc, 2025-03-20:
Thanks for investigating the cause of the large kernel resource usage. I changed the QtWebEngine memory allocation mechanism now to use a managed dataspace and AVL allocator for aligned allocations in `mmap()` instead of passing aligned random addresses to `mmap()` and did not see the kernel quota issues anymore so far.

alex-ab, 2025-05-19:
I'm closing the issue for now, we may reopen it if required.