genodelabs/genode #5468: base-hw: boot failures on platforms with sparse APIC IDs (Alder Lake+)

atopia, 2025-02-26:
During testing of #5215 we have encountered boot hangs on Alder Lake x86 platforms (e.g. Lenovo X1 Nano Gen 2, U7412 (fuji6)) which may stem from sparse APIC IDs. As noted in https://github.com/genodelabs/genode/issues/5215#issuecomment-2615405806, after reverting 9258004 and a7b4add the hangs did not occur.

cproc, 2025-03-19:
I appear to have the same problem with an i7-8650U CPU when hyper-threading is disabled. It works when hyper-threading is enabled or before commit 9258004.

This is the output of @chelmuth's patch from comment https://github.com/genodelabs/genode/issues/5310#issuecomment-2735707764:

hyper-threading disabled (bad case):

```
[-1] parse_lapic: cpu=0 acpi_id=1 apic_id=0                                                                                                                  
[-1] parse_lapic: cpu=1 acpi_id=2 apic_id=2                                                                                                                  
[-1] parse_lapic: cpu=2 acpi_id=3 apic_id=4                                                                                                                  
[-1] parse_lapic: cpu=3 acpi_id=4 apic_id=6
```

hyper-threading enabled (good case):

```
[-1] parse_lapic: cpu=0 acpi_id=1 apic_id=0                                                                                                                  
[-1] parse_lapic: cpu=1 acpi_id=2 apic_id=2                                                                                                                  
[-1] parse_lapic: cpu=2 acpi_id=3 apic_id=4                                                                                                                  
[-1] parse_lapic: cpu=3 acpi_id=4 apic_id=6                                                                                                                  
[-1] parse_lapic: cpu=4 acpi_id=5 apic_id=1                                                                                                                  
[-1] parse_lapic: cpu=5 acpi_id=6 apic_id=3                                                                                                                  
[-1] parse_lapic: cpu=6 acpi_id=7 apic_id=5                                                                                                                  
[-1] parse_lapic: cpu=7 acpi_id=8 apic_id=7
```

alex-ab, 2025-03-20:
The code in [repos/base-hw/src/bootstrap/spec/x86_64/platform.c](https://github.com/genodelabs/genode/commit/9258004cc7ff3bcf3af953c7a5ed17601500ed5f#diff-039b71ec66662944eee45a7b6e99034001934231f49d941d3d2577e0f27fd47f) line 348++ allocates the cpu memory dense, but with commit 9258004 the actually used [CPU IDs](https://github.com/genodelabs/genode/commit/9258004cc7ff3bcf3af953c7a5ed17601500ed5f#diff-75279e5db489c847521dafd34a0933e8557e87360bc8bfc9d70fa78c504fe3b9R77) are using directly the APIC IDs, which are not subsequent.

``` 
unsigned Bootstrap::Platform::_prepare_cpu_memory_area()
{
	using namespace Genode;

	for (size_t id = 0; id < board.cpus; id++)
		_prepare_cpu_memory_area(id);

	return board.cpus;
}
```

jschlatow, 2025-03-27:
Splendid! I cherry-picked your commits and am able to boot base-hw on the StarLite again.

chelmuth, 2025-03-28:
May I merge 03e19c9256b8a1743e50882ca7a8d316224bc65e and 455df9ab940db29785daed7595eabe595f43ffa5 to staging then?

cproc, 2025-03-28:
It works on my test machine too.

skalk, 2025-03-28:
> May I merge [03e19c9](https://github.com/genodelabs/genode/commit/03e19c9256b8a1743e50882ca7a8d316224bc65e) and [455df9a](https://github.com/genodelabs/genode/commit/455df9ab940db29785daed7595eabe595f43ffa5) to staging then?

@chelmuth that would be nice.

skalk, 2025-04-01:
@chelmuth I've re-worked the solution to circumvent the caching of available CPUs in a global array, which is not really needed, as @alex-ab rightly commented offline. The "fixup" commit is 94f6226. I'm sorry for the seesaw.

closed 2025-04-10 by nfeske