genodelabs/genode #5473: stdcxx: memory leak with 'std::mutex'

cproc, 2025-03-05:
`std::mutex` currently initializes a pthread mutex member with `PTHREAD_MUTEX_INITIALIZER` and does not call `pthread_mutex_destroy()`, so the libc-internal pthread mutex object which is allocated during the first `pthread_mutex_lock()` call is never freed. Possible solutions are to define `_GTHREAD_USE_MUTEX_INIT_FUNC`, which enables an alternative implementation which uses `pthread_mutex_init()` and `pthread_mutex_destroy()` instead of `PTHREAD_MUTEX_INITIALIZER`, or to add the `pthread_mutex_destroy()` call to the implementation with `PTHREAD_MUTEX_INITIALIZER`.

There already exists an open GCC bug report about this problem on FreeBSD [1] and I'm interested to find out if and how the problem has been solved for FreeBSD since the bug report creation in 2016.

[1] https://gcc.gnu.org/bugzilla/show_bug.cgi?id=71684

chelmuth, 2025-03-05:
Thanks for creating this issue (that I read right after sending the last mail ;-).

chelmuth, 2025-03-05:
It seems worthwhile to have a look into POSIX pthread_mutex_init *Static Initializers for Mutexes and Condition Variables* [1] that seems to distinguish *static* and *dynamically allocated* mutexes, with the latter be initialized by `pthread_mutex_init()`.

[1] https://pubs.opengroup.org/onlinepubs/9799919799/functions/pthread_mutex_destroy.html

chelmuth, 2025-03-05:
What about `pthread_cond_destroy()` and `_GTHREAD_USE_COND_INIT_FUNC`? `pthread_cond` objects are allocated via `new`.

cproc, 2025-03-05:
> What about `pthread_cond_destroy()` and `_GTHREAD_USE_COND_INIT_FUNC`? `pthread_cond` objects are allocated via `new`.

The implementation of `std::condition_variable` calls `pthread_cond_destroy()` even if `PTHREAD_COND_INITIALIZER` is used[1] and there is no `_GTHREAD_USE_COND_INIT_FUNC` macro like for `std::mutex`. I'm not sure why it is implemented differently.

[1] https://github.com/gcc-mirror/gcc/blob/8fc1a49c9312b05d925b7d21f1d2145d70818151/libstdc%2B%2B-v3/include/bits/std_mutex.h#L139

chelmuth, 2025-03-06:
Thanks for checking :+1:

cproc, 2025-03-19:
@chelmuth: the fix is relevant for drivers-24.10 as well since Qt6 was affected by the memory leak

chelmuth, 2025-03-20:
Thanks for the hint! I updated drivers-24.10 accordingly.

commit 7905c9e4 (2025-03-05): stdcxx: define _GTHREAD_USE_MUTEX_INIT_FUNC

Fixes #5473
closed 2025-04-10 by nfeske