genodelabs/genode #5475: system shell NOT in foreground

sevillalarry, 2025-03-08:
Hi Genode people,

I tried to follow the sculpt documentation up to "Tools -> system shell"
The "system shell" displayed on screen,BUT I canNOT type any commands.
when I click on the said shell/window, it canNOT be in the "foreground" or be an "active window".

Sources: https://genode.org/documentation/articles/sculpt-24-10
File QEMU/KVM: https://genode.org/files/sculpt/sculpt-24-10.img
File VirutalBox: https://genode.org/files/sculpt/sculpt-24-10.ova

Host A
Desktop: i7-8700 32Gb
OS: Gentoo GNU/Linux
Emulation: QEMU 9.2.0

Trial 1:
Note: -cpu remove due to error, -m 1024 too small, use ram fs
qemu-system-x86_64 -display sdl -machine q35 \
                    -m 6144 -drive format=raw,file=sculpt-24-10.img \
                    -netdev user,id=net0 -device e1000,netdev=net0

Trial 2:
Note: -cpu remove due to error, formated ahci-1.fs sculpt.qcow2 20Gb
qemu-system-x86_64 -display sdl -machine q35 \
                    -m 6144 -drive format=raw,file=sculpt-24-10.img \
                                  -drive format=raw,file=sculpt.qcow2 \
                    -netdev user,id=net0 -device e1000,netdev=net0

Host B
Desktop: i7-9700 32Gb
OS Windows 11
Emulation: VirtualBox 7.0.22

Trial: imported as Appliance sculpt-24-10.ova
VB guest hangs during adding Tools->system shell

Note: did NOT test on a desktop PC (no spare)

alex-ab, 2025-03-11:
The message seems to be identical to the one of the mailing list. Did you see the answer and followed the [suggestions](https://lists.genode.org/mailman3/hyperkitty/list/users@lists.genode.org/thread/QSPVNJWLIO3OJP552XFMIMQOWH2MU72R/)

nfeske, 2025-04-10:
The issue has become stale. I'm closing it.

closed 2025-04-10 by nfeske