genodelabs/genode #5476: Compiling demo issue

Kisel13, 2025-03-12:
I have started to get acquainted with the framework. I do everything according to the book, run `make run/demo KERNEL=linux BOARD=linux` and get this error:
```
  Program driver/framebuffer/sdl/fb_sdl
    COMPILE  main.o
In file included from /usr/include/SDL2/SDL_stdinc.h:33,
                 from /usr/include/SDL2/SDL_main.h:25,
                 from /usr/include/SDL2/SDL.h:31,
                 from /home/kisel/MyProj/KiselOS/genode/depot/genodelabs/src/fb_sdl/2025-02-26/src/driver/framebuffer/sdl/main.cc:29:
/usr/include/SDL2/SDL_config.h:54:2: error: #warning SDL_config.h might be incomplete, good luck [-Werror=cpp]
   54 | #warning SDL_config.h might be incomplete, good luck
      |  ^~~~~~~
cc1plus: all warnings being treated as errors
make[7]: *** [/home/kisel/MyProj/KiselOS/genode/repos/base/mk/generic.mk:73: main.o] Error 1
make[6]: *** [var/libdeps:290: fb_sdl.prg] Error 2
make[5]: *** [Makefile:351: gen_deps_and_build_targets] Error 2
make[4]: *** [/home/kisel/MyProj/KiselOS/genode/tool/depot/mk/build_bin_archive:214: /home/kisel/MyProj/KiselOS/genode/depot/genodelabs/bin/x86_64/fb_sdl/2025-02-26.build/bin] Error 1
make[3]: *** [/home/kisel/MyProj/KiselOS/genode/depot/var/build.mk:58: genodelabs/bin/x86_64/fb_sdl/2025-02-26] Error 2
make[2]: *** [/home/kisel/MyProj/KiselOS/genode/tool/depot/build:159: execute_generated_build_mk_file] Error 2
make[1]: *** [/home/kisel/MyProj/KiselOS/genode/tool/depot/create:59: build] Error 2

Error: tool/depot/create during depot-auto-update failed

make: *** [Makefile:446: run/demo] Error 1
```

rite, 2025-03-12:
I've seen this too (on Arch Linux). Apparently, it is related to https://github.com/libsdl-org/sdl2-compat/commit/f4abaea268893ad8b67acdb732d8cd20b3072feb, which has been released recently. I can't provide a proper fix though, I currently just work around the issue by commenting out the the warning in `/usr/include/SDL2/SDL_config.h` 😳 ;)

Kisel13, 2025-03-12:
I'm also using Arch Linux, maybe I should try installing an earlier version of the package

Kisel13, 2025-03-12:
This error went away after changing the config, thank you. In the meantime, we'll wait for updates, maybe they'll fix it then.

atopia, 2025-03-13:
Thanks for reporting this issue and thank you @rite for finding the cause for this regression. I have reproduced this issue on Arch Linux and [64c556f](https://github.com/atopia/genode/commit/64c556fda702c34b4fe9e84a15519594e311cb32) fixes the issue for me.

closed 2025-04-06 by Kisel13