genodelabs/genode #5477: vbox6: race condition in multi-monitor-support

atopia, 2025-03-12:
Commits 2cdcacf2 and 71e0fa201 introduce or expose a race condition in the initialization of the virtual display in an automated setting, which on nova manifests in an empty black Virtualbox window and on hw leads to no window.

### Success rate

| test | success rate |
| ----- | ---- |
| 71e0fa201, nova  | 40% success, 10 tries |
| 71e0fa201, hw  |  0% success, 5 tries |
| cec3a824013b5b3, nova |  100% success, 10 tries | 
cec3a824013b5b3, hw| 100% success, 10 tries|

chelmuth, 2025-03-26:
@atopia please test 2ffbf05e3c46d959682220eeb0689e8566537e37 in your scenario

atopia, 2025-03-26:
In some first testing the fix on top of staging did not trigger the error on nova (worked 3/3 times). For maybe unrelated reasons the scenario didn't work on base-hw. I will give it some more testing tomorrow.

atopia, 2025-03-27:
My branch [5477-test-vbox-fix-reduced](https://github.com/atopia/genode/tree/5477-test-vbox-fix-reduced) works via the reverted commits on top, the state at f4deef070a23c47990487f5c33dbba66beec007e fails with my scenario on base-hw (but works on nova so far) despite containing your fix.

chelmuth, 2025-03-27:
In my case (U7411, tinycore guest) it works until X.org comes up. Unfortunately, screen updates happen only on VT changes in X, while the console works flawlessly.

atopia, 2025-03-31:
FYI, not the base-hw scenario restarting the vbox component does not resolve the issue.

closed 2025-04-10 by nfeske