genodelabs/genode #5479: Turn part_block into a mediator to stream-line I/O access

cnuke, 2025-03-13:
`part_block` allows for accessing each partition on a block-provider, normally a block device, individually. It sits in-between the block-provider component and the components requiring access, e.g. the file-system provider. Each client gets its own *Block* session covering the range of blocks belonging to the particular partition while `part_block` establishes one session to interact with the provider component itself that covers the whole range of blocks.

Structuring it in this way separates all participants but at the same time results in having to deal with the payload twice because two Packetstreams are involved on the I/O path. Depending on the use-case at hand that binds resources, i.e. CPU cycles, that could be spend on other tasks.

Following the example of the `chroot` component, which mediates the *File_system* session access, combined with the recent change of limiting the scope via the *File_system* session's label, could also pave the way for turning `part_block` around in a comparable fashion.

Annotating the *Block* session with a start offset and range of blocks (or similar information) covered by the session arguments would allow for selecting a specific portion of the block-provider and by allowing the client to open the session directly at the source one hop in the I/O path could be spared. As a side-effect the settings of each session can be tailored more closely to each particular use-case.

As a down-side this pushes more complexity into the block-provider and depending on the driver the benefits might be limited (in case where there is only one useable queue with a more constrainted number of requests) and softens the separation.

cnuke, 2025-03-20:
The referenced commits above illustrate the idea, while `vfs_block` is currently the only block-provide allowing for mutliple concurrent clients.

cnuke, 2025-03-27:
Here a some results running `fio` (version 3.39) on Genode that compare the *copying* with the *mediating* `part_block` component based on my WIP [drivers-24.10-disk_optimization-2025-03-25](https://github.com/cnuke/genode/commits/drivers-24.10-disk_optimization-2025-03-25/) branch.

The configurations are as follows:

* *iops-read* uses randread with an I/O depthqueue of 32, a blocksize of 4096 bytes and spawns 4 jobs
* *iops-write* uses randwrite with an I/O depthqueue of 32, a blocksize size of 4096 bytes and spawsn 4 jobs
* *latency-read* uses read with an I/O depthqueue of 1, a blocksize of 4096 bytes and spawns 1 job
* *latency-write* uses write with an I/O depthqueue of 1, a blocksize of 4096 bytes and spawns 1 job
* *throughput-read* uses read with an I/O depthqueue of 1024 KiB and spawns 1 job
* *throughput-write* uses write with an I/O depthqueue of 1024 KiB and spawns 1 job

All use the *sync* I/O engine as other ones are currently not available, which always caps the I/O depthqueue to 1, and have a fixed duration of 30 seconds.

The `vfs/block` plugin is configured with a bulk-size of 4 MiB and `driver/nvme` as well as `server/vfs_block` make of multiqueue operations when used by the *mediated* `part_block`.

Notes:

* The first number denotes the bandwith (*MiB/s*) while the second the I/O operations.
* The *vfs/ram* variant does not make use of `part_block` and is included as a base-line and to some degree show-cases the variances.
* Both *nvme* and *vfs_block* use the `vfs/block` plugin to access `part_block`.
* The CPU load generated by each test is attached in the details section and should be consulted to put the results into perspective.
* The system machine contains a Intel i7-1185G7 CPU.

<hr>

*copying* `part_block`, ❶-❸ are running `fio` on a single core while ❹-❻ use two cores

|   | block     | iops-read      | iops-write     | latency-read  | latency-write  | throughput-read | throughput-write |
| - | -         | -              | -              | -             | -              | -               | -                |
| ❶ | nvme      |  25.1   (6423) |  54.3  (13.9k) |  50.9 (13.0k) |  83.0  (21.2k) |  1136.0  (1136) |  850.0  (850)    |
| ❷ | vfs_block | 155.0  (39.7k) | 173.0  (44.3k) | 119.0 (30.6k) | 153.0  (39.3k) |  5316.0  (5316) | 3815.0 (3815)    |
| ❸ | vfs/ram   | 671.0 (172.0k) | 821.0 (210.0k) | 240.0 (61.6k) | 461.0 (118.0k) | 11800.0 (12.1k) | 5842.0 (5841)    |
| - | -         | -              | -              | -             | -              | -               | -                |
| ❹ | nvme      |  24.8   (6337) |  53.1  (13.6k) |  50.7 (13.0k) |  81.5  (20.9k) |  1088.0  (1087) |  844.0  (843)    |
| ❺ | vfs_block | 150.0  (38.3k) | 173.0  (44.3k) | 127.0 (32.4k) | 149.0  (38.2k) |  5237.0  (5236) | 3515.0 (3514)    |
| ❻ | vfs/ram   | 452.0 (116.0k) | 748.0 (191.0k) | 264.0 (67.5k) | 482.0 (123.0k) | 12000.0 (12.3k) | 5782.0 (5781)    |

<details>
  <summary>top for results in ❶</summary>

<details>
  <summary>iops-read</summary>

```
[init -> top] cpu=0.0   0     0  82.50%  83.39% thread='idle'                    label='kernel'
[init -> top] cpu=0.0  64 10000   5.67%   0.00% thread='ep'                      label='init -> nvme'
[init -> top] cpu=0.0  64 10000   4.63%   0.00% thread='ep'                      label='init -> part_block'
[init -> top] cpu=0.0  64 10000   1.22%   5.96% thread='part_block'              label='init -> part_block'
[init -> top] cpu=0.0  64 10000   1.09%   1.12% thread='signal handler'          label='init -> part_block'
[…]
[init -> top] cpu=1.0   0     0  91.57%  91.45% thread='idle'                    label='kernel'
[init -> top] cpu=1.0  64 10000   4.91%   0.00% thread='ep'                      label='init -> sequence -> fio'
[init -> top] cpu=1.0  64 10000   0.82%   0.84% thread='signal handler'          label='init -> sequence -> fio'
[init -> top] cpu=1.0  64 10000   0.67%   0.67% thread='pthread.5'               label='init -> sequence -> fio'
[init -> top] cpu=1.0  64 10000   0.67%   0.68% thread='pthread.6'               label='init -> sequence -> fio'
[init -> top] cpu=1.0  64 10000   0.67%   0.67% thread='pthread.4'               label='init -> sequence -> fio'
[…]
```
</details>

<details>
  <summary>iops-write</summary>

```
[init -> top] cpu=0.0   0     0  64.09%  64.84% thread='idle'                    label='kernel'
[init -> top] cpu=0.0  64 10000  12.25%   0.00% thread='ep'                      label='init -> nvme'
[init -> top] cpu=0.0  64 10000   9.87%   0.00% thread='ep'                      label='init -> part_block'
[init -> top] cpu=0.0  64 10000   2.66%  17.19% thread='nvme'                    label='init -> nvme'
[init -> top] cpu=0.0  64 10000   2.63%  12.76% thread='part_block'              label='init -> part_block'
[init -> top] cpu=0.0  64 10000   2.35%   2.43% thread='signal handler'          label='init -> part_block'
[... 1473 lines cut ...]
[init -> top] cpu=0.0  64 10000   0.61%  47.48% thread='vfs_block'               label='init -> vfs_block'
[init -> top] cpu=0.0  64 10000   0.41%   0.00% thread='ep'                      label='init -> timer'
[…]
[init -> top] cpu=1.0   0     0  50.21%  50.16% thread='idle'                    label='kernel'
[init -> top] cpu=1.0  64 10000  30.65%   0.00% thread='ep'                      label='init -> sequence -> fio'
[init -> top] cpu=1.0  64 10000   9.41%   9.43% thread='pthread.6'               label='init -> sequence -> fio'
[... cut ...]
[init -> top] cpu=2.0  64 10000   8.93%   9.04% thread='pthread.3'               label='init -> sequence -> fio'
[init -> top] cpu=2.0   0     0   0.00%   0.00% thread='pager'                   label='core'
[init -> top] cpu=2.0   0     0   0.00%   0.00% thread='killed'                  label='kernel'
[init -> top] cpu=2.0   0     0   0.00%   0.00% thread='cross'                   label='kernel'
[…]
```
</details>
</details>

<details>
  <summary>top for results in ⑥ (empty - similar to ❻)</summary>

</details>

chelmuth, 2025-03-28:
I see two takeaways here.

1. The mediator reduces CPU load in the nvme throughput test by 15%.
2. I did not expect *top for results in ③ (empty - similar to ❸)*. Do you have an explanation why the CPU-load reduction does not impact this scenario?

cnuke, 2025-08-11:
The following commits addresses remarks from the first review and short-comings made apparent while tackling those.

Note that most drivers do not query the medium state, i.e. if the medium is *writeable* or *read-only*, and thus reporting `writeable=true` for the *physical* session info is done for allowing the check on session creation to proceede (the `Request_stream` is given the *physical* information and the constraining view to produce the *virtual* information the client is using). In the same vein, the
`driver/usb_block` does not evaluate *policy* nodes so far, which makes the `writeable` attribute given in its config somewhat obsolete. 

* c04c7f0585 - fixup 'block_session: fix partial reads in Job API' (consumed_count) - changes the naming to be more precsice

* e90cdbcd2b - fixup 'block_session: add constraining view type' (check write allowance) - fixes a missing check that checks the write allowance for all request types where this relevant at the request-stream level (and thereby can be removed from the drivers).

* ffdeeb02a8 - fixup 'ahci: implement SQ multiplexing' (check writeable in policy) -  adapts to the previous commit write allowance commit.
* 77fb63e485 - fixup 'genode_c_api/block: implement SQ multiplexing' (writeable from policy) -  adapts to the previous commit write allowance commit.
* bb942465d8 - fixup 'usb_block: implement SQ multiplexing' (writeable check in request_stream) -  adapts to the previous commit write allowance commit.
* f4b87da2c8 - block: add Constrained_view::from_args utility -  adds a helper to creates the view object from the args string and removes the duplicated to code from the components where appropriate.
* d68a0f6f33 - driver/usb_block: reorder request handler -  changes the ordern in which the various lambda functions are spread out in the request handler.
* 44fc34d6db - driver/ahci: reorder request handler -  changes the ordern in which the various lambda functions are spread out in the request handler.
* a87db8140e - driver/nvme: reorder request handler -  changes the ordern in which the various lambda functions are spread out in the request handler.
* 673fef74ae - vfs_block: return view as copy
* 4bda213c13 - driver/ahci: solely use id to free DMA buffer
* b6f22fe496 - driver/nvme: remove writeable function

nfeske, 2025-08-28:
Featured in Genode 25.08

closed 2025-08-28 by nfeske