genodelabs/genode #5483: hw: assert cpu core in `run_vm` and `pause_vm` call, simplify initialization

skalk, 2025-03-18:
As discussed in issue #5474 currently the initialization of hardware controlled virtualization state buffers for Intel VMX is done in a complicated fashion. It has to be assured that those hardware buffers are initialized and accessed on the target cpu only. On the other hand it is obviously not assured that base-hw kernel calls, which should access these buffers to synchronize its values with a VMM client, are only called by threads running on the right target cpu core. Therefore I propose to:

1. Add an error message in run/pause calls when the wrong cpu core was used, and ignore the action
2. Change the base-hw base-library parts that call pause/run, so only the right entrypoints use them
3. Simplify the hardware buffer (VMCS correct term?) initialization to not first schedule a Vcpu and do it there, but to initialize it when run is called the first time

atopia, 2025-03-24:
@skalk please have a look if this is what you had in mind. I've successfully tested the change with `run/vmm_x86` on qemu/SVM and Intel hardware and with the Virtualbox scenario on my Intel test machine.

skalk, 2025-03-25:
@atopia nice, yes that matches exactly what I had in my mind. The only thing left is setting this exit code stuff in the construction within the base-library, and remove that part from the kernel too. But this can be addressed in a separated commit too. From my side, this commit can move to staging.

atopia, 2025-03-25:
@skalk commit [4c4b518](https://github.com/atopia/genode/commit/4c4b518aad41a0c722be0b31ecc9a7611fa3fa97) moves setting startup exit from the kernel to the VMM library and commit [fd80f8e](https://github.com/atopia/genode/commit/fd80f8e50541885004f5eedc198a96cc4e083005) remedies potentially confusingly named methods around reading and writing the vCPU state.

atopia, 2025-03-25:
As discussed offline [e87ddb5](https://github.com/atopia/genode/commit/e87ddb5f49ad383b68630ae21110d90a38473fec) renames the `*_vcpu_state` methods to `load()` and `store()`.

skalk, 2025-03-27:
@chelmuth green light from my side for both commits

commit 0b269788 (2025-03-25): hw: simplify x86 vCPU initialization

Initializing a vCPU on base-hw involved a round-trip to the kernel to
send the initial startup exit signal and setting up the vCPU on first
run of the vCPU thread.

Signal the vCPU handler directly from the base libary. This removes the
need to call `Vm::run()` during construction of the vCPU, possibly from
a remote core. Check that `Vm::run()` and `Vm::pause()` are only called
from the local CPU core. Finally, move the initialization of the vCPU
from `Vm::proceed()` to `Vm::run()`, so that the `Vcpu_state` can be
synced from the VMM directly after initial [... cut]
commit 7a03cd37 (2025-03-27): fixup! hw: simplify x86 vCPU initialization

Set the STARTUP_EXIT from the Hw_vcpu constructor and move its
definition out of the kernel. Check the vCPU's initialization state on
x86_64 at the level of the `pause()` system call and remove the
`_sync_from_vmm()` and `_sync_to_vmm()` hooks that are superseeded by
specializing `pause()` for x86_64.

Fixes #5483
closed 2025-04-10 by nfeske