genodelabs/genode #5485: sculpt_manager: popup dialog closes inadvertently when controlled with touch events

jschlatow, 2025-03-18:
When testing Sculpt on the StarLite tablet, I noticed that the popup dialog always closes when I try to touch anything within the dialog. This is basically due to the closing logic not waiting for the updated hover report.

jschlatow, 2025-03-18:
@nfeske commit d397bc6 adds sequence numbers to nitpicker's hover report and lets the sculpt manager evaluate this report to identify whether the popup dialog was clicked/touched as we have discussed offline.

jschlatow, 2025-03-18:
I just noticed that the commit does not seem to reliably solve this issue. Sometimes, the popup dialog still closes unexpectedly.

jschlatow, 2025-03-18:
@nfeske Please have a look at commit c245ce7, which solves this issue in a more robust manner.

nfeske, 2025-03-19:
Practical stress-testing notwithstanding, your commit looks very good to me. I'm happy that the approach of using the hover report worked out as I was hoping for. :) I merged your commit to staging.

nfeske, 2025-03-19:
BTW, on staging you can also find a sequence of multi-monitor-related fixes that address corner cases (like vertically moving the pointer along x=0) identified prior and during the exhibition last week.

closed 2025-04-10 by nfeske

nfeske, 2025-04-11:
@jschlatow and @ssumpf, I think that I have found a clean solution for the popup and touch handling. The solution (on staging) consists of a clean separation of touch from pointer input by nitpicker, the same for the sculpt manager, and the interpretation of touch-release by menu-view. Note that I have tested this only in my artificial test scenario on base-linux, not of PC hardware or PinePhone yet. Testing from your side would be very welcome.

nfeske, 2025-04-11:
By further testing with a pointer device, I already noticed one glitch: After clicking into the popup once, the popup closes the pointer leaves its area.

nfeske, 2025-04-11:
I have just fixed this corner case on staging.

ssumpf, 2025-04-14:
@nfeske: It works as intended on the PinePhone.

nfeske, 2025-04-14:
@ssumpf thank you for test-driving! 👍

jschlatow, 2025-04-14:
@nfeske I tested your changes on the StarLite. Touch-interactions with the panel dialog seem to work. However, after touch-opening the popup dialog and touching once into it, it closed. Apparently, this broke ever interaction (touch or pointer) with the runtime dialog. I'm also getting A LOT of "more than one dialog hovered" messages from menu view.

jschlatow, 2025-04-14:
Another observation (before the situation described above): When I hover the Log-Button with the pointer, touching buttons in the runtime graph are interpreted as touches on the Log-Button.

nfeske, 2025-04-14:
Thanks @jschlatow for the feedback.

> I'm also getting A LOT of "more than one dialog hovered" messages from menu view.

I saw these messages, too. They should have been addressed by b540716. I have seen your comment about the quickly consecutive touch/touch-release combo. I'll need to think about how to best handle that. Maybe, the "multiple hovered" dialogs could be disambiguated by tracking which dialog received input most recently? 

> When I hover the Log-Button with the pointer, touching buttons in the runtime graph are interpreted as touches on the Log-Button.

That is most probably caused by the dialog framework detection of pointer vs. touch. The combination of both is currently not considered there.

jschlatow, 2025-04-14:
> Another observation (before the situation described above): When I hover the Log-Button with the pointer, touching buttons in the runtime graph are interpreted as touches on the Log-Button.

I believe this is a direct consequence from the "more than one dialog unexpectedly hovered at the same time"-Warnings: The Log-Button is kept hovered because the pointer never left the button. Additionally, another dialog gets hovered because of a touch event.  @nfeske I think menu_view should track the sequence number with the hover state for each dialog and report the hover state for the dialog with the most recent sequence number. Does this sound like a reasonable approach?

By the way, I am also wondering about the expected behaviour of the following input sequence:
1. Move the mouse pointer over a button (e.g. Log-Button).
2. Touching another button.
3. Emitting a single mouse click (without moving the pointer).

Since the pointer still visually hovers the Log-Button, the user might expect to be able to click it without moving the pointer. I do not see a way to make clicks effective in the described situation. May it be better to make the hover state global in menu view so that touching elsewhere steals the hover state from the previously hovered element? Should this be solved in menu_view at all, or do you think it would be better for nitpicker to invalidate the pointer state when a touch event occurs?

jschlatow, 2025-04-14:
> The explanation for this behaviour is that there are multiple touch reports generated for the same sequence number. The last report (just before the popup dialog closes) does not state any session so that it is interpreted as a touch into nowhere. I believe this is because the dialog content changed in a way that nitpicker unhovered the client. I ran into this before and solved it by only considering the first report for each sequence number.

Commit f89d965 should fix this. The popup dialog is still not working properly (after closing it intentionally I'm unable to open it again). Yet, this is probably related to the hover-report issue from menu_view.

nfeske, 2025-04-14:
@jschlatow could you please give my new take (on staging) a try?

jschlatow, 2025-04-15:
> [@jschlatow](https://github.com/jschlatow) could you please give my new take (on staging) a try?

It solves the Log-Button issue. I am delighted that even a single click (without pointer movement) works perfectly after I touch-controlled the graph.

There are a few remaining minor corner cases, though:
1. Touching the "Add" button or selecting an option in the popup dialog closes the dialog.
2. After touch closing the popup dialog (either intentionally or unintentionally), clicks are ineffective. Click-functionality is restored after touching any button.

I'll have a go at adapting my previous solution for 1 to the new popup-close logic.

jschlatow, 2025-04-15:
> 1. Touching the "Add" button or selecting an option in the popup dialog closes the dialog.

Fixed by 8242695 and 12e0191.

> 3. After touch closing the popup dialog (either intentionally or unintentionally), clicks are ineffective. Click-functionality is restored after touching any button.

Still a mystery. Interestingly, though, the hover effect is visible when moving the pointer. Investigating...

nfeske, 2025-04-15:
Thank you for the fixups, which I merged to staging just now.

> Still a mystery. Interestingly, though, the hover effect is visible when moving the pointer. Investigating...

Minor bug fixes not withstanding, would you be fine with deferring the mixed use of touch and pointer at large to a point after the Sculpt release? I sense that we need to think through this topic before approaching a lasting solution. May the Hack'n'Hike be a good opportunity for that?

jschlatow, 2025-04-15:
> Minor bug fixes not withstanding, would you be fine with deferring the mixed use of touch and pointer at large to a point after the Sculpt release? I sense that we need to think through this topic before approaching a lasting solution. May the Hack'n'Hike be a good opportunity for that?

Yes, that is a good option.

commit 235a38b2 (2025-03-19): sculpt_manager: touch control of popup dialog

When trying to control the popup dialog with touch events, the dialog
was immediately closed (on any touch). This was a consequence of
evaluating the dialog's hover state without waiting for the
corresponding hover report. For regular motiong events, the hover report
is updated when the pointer moves. For touch events, however, the
hover report is only updated when the touch occurs. We therefore need to
wait for the hover report that corresponds to the touch event before
deciding about closing the popup dialog.

Since menu_view's hover report is n [... cut]