genodelabs/genode #5488: fb_sdl error on Xubuntu 24.04

cproc, 2025-03-25:
When I run the `framebuffer` test on Xubuntu 24.04, the following error occurs:

```
[init -> test-framebuffer] --- Test framebuffer ---
Kernel: Error: Missing call of init_log
```

GDB shows the following backtrace for `fb_sdl`:

```
#0  pseudo_end () at /.../repos/base-linux/src/lib/syscall/spec/x86_64/lx_syscall.S:29
#1  0x000000000042c35e in lx_futex (val=0, op=0, uaddr=<optimized out>) at /.../repos/base-linux/src/lib/syscall/linux_syscalls.h:432
#2  thread_stop_myself (myself=<optimized out>) at /.../repos/base-linux/src/include/base/internal/lock_helper.h:62
#3  Genode::Lock::lock (this=this@entry=0x7fffffffeb10, myself=...) at /.../repos/base/src/lib/base/lock.cc:128
#4  0x000000000042c41f in Genode::Lock::lock (this=this@entry=0x7fffffffeb10) at /.../repos/base/src/lib/base/lock.cc:66
#5  0x0000000000434005 in Genode::Blockade::block (this=0x7fffffffeb10) at /.../repos/base/include/base/blockade.h:41
#6  Genode::sleep_forever () at /.../repos/base/src/lib/base/sleep.cc:21
#7  0x000000000043c178 in Genode::Log::log () at /.../repos/base/src/lib/base/default_log.cc:65
#8  0x00000000004424ca in Genode::error<char const (&) [134]> () at /.../repos/base/include/base/log.h:184
#9  operator delete () at /.../repos/base/src/lib/cxx/new_delete.cc:83
#10 0x00007fffee04da2f in ?? () from /lib/x86_64-linux-gnu/libLLVM.so.19.1
#11 0x00007fffee04d915 in ?? () from /lib/x86_64-linux-gnu/libLLVM.so.19.1
#12 0x00007fffee04d3c9 in ?? () from /lib/x86_64-linux-gnu/libLLVM.so.19.1
#13 0x00007fffedd045dd in ?? () from /lib/x86_64-linux-gnu/libLLVM.so.19.1
#14 0x00007ffff7fca71f in call_init (l=<optimized out>, argc=argc@entry=1, argv=argv@entry=0x7fffffffeda8, env=env@entry=0x7fffffffedb8) at ./elf/dl-init.c:74
#15 0x00007ffff7fca824 in call_init (env=<optimized out>, argv=<optimized out>, argc=<optimized out>, l=<optimized out>) at ./elf/dl-init.c:120
#16 _dl_init (main_map=0x7ffff7ffe2e0, argc=1, argv=0x7fffffffeda8, env=0x7fffffffedb8) at ./elf/dl-init.c:121
#17 0x00007ffff7fe45a0 in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#18 0x0000000000000001 in ?? ()
#19 0x00007fffffffef7c in ?? ()
#20 0x0000000000000000 in ?? ()
```

So, apparently `libLLVM` calls an unimplemented `delete` operator in the `cxx` library. When I replaced `Genode::error()` with `Genode::raw()` in the dummy implementations I saw that different `delete` operators got called multiple times, but other than that the test worked. Building SDL2 from source solved the problem as well, probably because `ldd` did not show `libLLVM` as dependency anymore.

chelmuth, 2025-05-06:
@cproc please test current staging, which includes the fixes of #5560.

cproc, 2025-05-06:
> [@cproc](https://github.com/cproc) please test current staging, which includes the fixes of [#5560](https://github.com/genodelabs/genode/issues/5560).

It's working now.

closed 2025-05-06 by cproc