genodelabs/genode #5489: Modernize block-tester implementation

nfeske, 2025-03-25:
In the context of #5411, I came across the block tester, which calls for an update of its internal coding style and structure.

nfeske, 2025-03-25:
Done by commit https://github.com/nfeske/genode/commit/c562eb0f90cedf90871e30a5ca3bb244994c64aa. @cnuke may you give this a look/try at your convenience? ;-)

cnuke, 2025-03-27:
@nfeske the changes LGTM and are very much appreciated. I've mostly cosmetic additions in commit [92bb00e](https://github.com/cnuke/genode/commit/92bb00ee83f22dd79c768b518a8a70dff860b61f) but it also restores *mibs* and *iops* in the LOG report. I think those should be added unconditionally.

nfeske, 2025-03-27:
Thank you very much @cnuke for the review. Your final touches are highly appreciated. Merged to staging.

chelmuth, 2025-03-28:
It seems Number_of_bytes` does not fit use for block-related numbers/counts on 32-bit platforms as it uses `size_t`.

```
> make app/block_tester
...
.../genode/staging.git/repos/os/src/app/block_tester/test_random.h:168:63: error: conversion from ‘Genode::uint64_t’ {aka ‘long long unsigned int’} to ‘Genode::size_t’ {aka ‘long unsigned int’} may change value [-Werror=conversion]
  168 |                                    "length:", Number_of_bytes(_length), " ");
```

cnuke, 2025-03-28:
Sorry, that's on me. Commit [a0d3adc](https://github.com/cnuke/genode/commit/a0d3adc1849b920391f882b78aed03b109bc9245) addresses the issue by providing a new type expressing the intent.

nfeske, 2025-03-28:
Commit https://github.com/genodelabs/genode/commit/c8b7af8c504a3e6899bbc66261addc6776d8d2da makes the formatted output feature of Number_of_bytes utility reusable.

Commit https://github.com/genodelabs/genode/commit/e587f5683f4e4b5c9323da840680ec62139c3945 wraps the basic types used by the block tester, with the formatted output added to the `Total` type.

@cnuke do you approve?

cnuke, 2025-03-28:
@nfeske yes :+1:

commit 904b4e72 (2025-03-27): block_tester: modernize internal structure

Don't copy Xml_node, remove the use of exceptions, split test mechanics
from the scenario's data representation, avoid inheritance, use const
where possible.

Note that this patch also changes the batching of the <replay> scenario.
Unlike the other scenarios, it did not use to cap the maximum number of
jobs according to 'batch' attribute. The interpretation of the 'batch'
attribute has now been unified.

Issue #5411
Fixes #5489
closed 2025-04-10 by nfeske