genodelabs/genode #5501: Sculpt OS 25.04 - PC

cnuke, 2025-04-01:
This issues documents the progress of the upcoming Sculpt-OS release for PCs.

cnuke, 2025-04-01:
I made a testing image (*2025-04-01*) and the corresponding depot archives available that contains @alex-ab's fix ([7fc0c6b](https://github.com/alex-ab/genode/commit/7fc0c6bd7001d1c473225a8c7adac6bcb358f122)) as well as @ssumpf's dialog related commit [3649444](https://github.com/ssumpf/genode/commit/3649444f855efec8c9273377deac1b9a4700ca48). It is build with `LOG=core` and contains the unaltered presets that may reference outdated archives (e.g. *falkon* and *goa_testbed*).

I spent the afternoon to no avail trying to fix the deployment of the *fonts_fs* pkg that is not able to access its configuration—it always gets `<config/>`, which leads to `Error: VFS not configured`. I briefly instrumented `config_fs_rom` but I see no requests from *fonts_fs*.

chelmuth, 2025-04-02:
If a monitor larger then the notebook internal display is connected on boot or plugged later, the screen stays black und the log shows *Warning: surface 2560x1440 mismatches texture 1920x1080* repeatedly. This is somewhat related to https://github.com/genodelabs/genode/issues/5428#issuecomment-2761304169 but could also be addressed in intel_fb explicitly.

chelmuth, 2025-04-02:
> If a monitor larger then the notebook internal display is connected on boot or plugged later, the screen stays black und the log shows _Warning: surface 2560x1440 mismatches texture 1920x1080_ repeatedly. This is somewhat related to [#5428 (comment)](https://github.com/genodelabs/genode/issues/5428#issuecomment-2761304169) but could also be addressed in intel_fb explicitly.

@atopia the issue you mentioned via email may be just this. Could you try to boot with no external displays connected?

atopia, 2025-04-02:
I have, but to no avail. I did then realize that the X1 Nano's internal display has a resolution of 2160x1350, so I'm suspecting  #5428.

chelmuth, 2025-04-02:
> I spent the afternoon to no avail trying to fix the deployment of the _fonts_fs_ pkg that is not able to access its configuration—it always gets `<config/>`, which leads to `Error: VFS not configured`. I briefly instrumented `config_fs_rom` but I see no requests from _fonts_fs_.

Fixed by ae2e4cbc3a2fd58dc7a494aba85b43f8b13c045c.

alex-ab, 2025-04-02:
> If a monitor larger then the notebook internal display is connected on boot or plugged later, the screen stays black und the log shows _Warning: surface 2560x1440 mismatches texture 1920x1080_ [...]

@chelmuth: please try 290de2b379735283ea79cdb5e79c9889b528bec9, works here with my test setup

nfeske, 2025-04-02:
Commits https://github.com/genodelabs/genode/commit/4fe87c0134ab207bf37430e797ab6e02e75058de and https://github.com/genodelabs/genode/commit/ad8768d35ec210df2afbb1bfc603063a57aca33c should hopefully solve the black-screen issue with display resolutions that are not a multiple of 8x8 pixels. @jschlatow and @atopia I would appreciate you testing these commits with your machines. With these changes, the back2front operation falls back to a simple per-pixel copy in the non-8x8 case, instead of backing out. For a non-rotated screen on a PC, this should not make much of performance difference.

atopia, 2025-04-02:
The commits don't fix Sculpt with Nova on the X1 Nano for me (neither on staging nor on 25.02 with the commits picked). As I simply don't get past the Genode boot screen without an easy way to debug the setup, I wouldn't put too much weight on this, it was worth a try but the source of the problem could be something else entirely.

jschlatow, 2025-04-02:
@nfeske The commits do fix Sculpt/NOVA on the x230 (tested on staging).

atopia, 2025-04-07:
My boot problems on the X1 Nano appear to be caused by a local build misconfiguration, a test image provided by @chelmuth worked just fine.

cnuke, 2025-04-07:
I noticed a visible pattern in the *leitzentrale* on my X280 (with a touchscreen): [image]

I can also reproduce that on the Fuji5b (U7411) (also with a touchscreen) but not on a Fuji4 (S938) or Fuji3 (S937). However, the test3-x201 (w/o a touchscreen) also shows the pattern but in contrast to the other machines there it is barely visible.

Booting 24.10 on the Fuji5b and test3-x201 does not exhibit the pattern.

I have an [image](https://depot.genode.org/cnuke/image/sculpt-pc-2025-04-04-1743.img.xz) built on friday that can be used if somebody wants to check their test machines for this behavior as well. Or, to exclude issues solely existing within my image, of course you can use your own.

cnuke, 2025-04-07:
Using a touchscreen leads to non-hovered buttons on 25.04 (tested with X280 and Fuji5b (U7411) in contrast to 24.10 where the widgets are hovered. Both versions have in common that the mouse pointer is offset to the bottom right by a noticeable margin (a few hundered pixels) from where the finger touched the display.

(I have no prior experience with using a touchscreen with Sculpt with a Laptop so I am not sure what to expected.)

cnuke, 2025-04-07:
> I tested on U7411 without touchscreen (`display_name` in _report/runtime/intel_fb/connectors_ is `CMN14E4`) and the pattern is almost invisible. An external monitor does not show this effect. @cnuke what's your `display_name`?

@chelmuth it is `LGD0660`.

jschlatow, 2025-04-07:
> I noticed a visible pattern in the _leitzentrale_ on my X280 (with a touchscreen): [image]
>
> I can also reproduce that on the Fuji5b (U7411) (also with a touchscreen) but not on a Fuji4 (S938) or [...]

 I can reproduce this pattern on a x260 and the StarLite. It is barely visible on the internal displays (only when you know what you're looking for) with an external display on full brightness, however, it becomes clearly visible.

chelmuth, 2025-04-07:
> I can reproduce this pattern on a x260 and the StarLite. It is barely visible on the internal displays (only when you know what you're looking for) with an external display on full brightness, however, it becomes clearly visible.

We identified commit 146f161383d8d488dee44279d370ad18d8676427 (gui_fader) as the culprit. Investigating...

cnuke, 2025-04-07:
> @cnuke Could provide more details: What buttons are not hovered?

If I move the mouse pointer via the touchscreen over any `menu_view` item in the runtime graph, e.g. `intel_fb`, the button is not outlined (thick black border). As soon as afterwards I move  an external USB mouse device the border appears.

> What touchscreen driver do you use? Any particular event-filter config?

It uses the `usb_hid` driver on the Fuji5b (I don't have the X280 at hand) and it is the unaltered default configuration generated by Sculpt.

```
[…]
[init -> runtime -> usb_hid] input: Melfas LGDisplay Incell Touch as /devices/usb-1-8/0-0:1.0/0003:1FD2:9003.0002/input/input1
[init -> runtime -> usb_hid] Connected device: input1 (Melfas LGDisplay Incell Touch at usb-usbbus-0/input0) MULTITOUCH TOUCHSCREEN
[init -> runtime -> usb_hid] hid-multitouch 0003:1FD2:9003.0002: input: USB HID v1.11 Device [Melfas LGDisplay Incell Touch] on usb-usbbus-0/input0
[init -> runtime -> usb_hid] hid-generic 0003:1FD2:9003.0003: device has no listeners, quitting
[…]
```

> You could add a `<log motion="yes">` node to the event filter (around the touch-input source) and touch the display corners? This would help identify whether the touch-event coordinates are already offset or whether this happens later in the processing chain.

Touch at the upper left corner:

```
[init -> event_filter] Input event #3795        TOUCH_RELEASE 0 key count: 0
[init -> event_filter] Input event #3796        TOUCH 0 +2.0+0.0        key count: 0
[init -> event_filter] Input event #3797        TOUCH 0 +3.0+0.0        key count: 0
[init -> event_filter] Input event #3798        TOUCH 0 +4.0+0.0        key count: 0
[init -> event_filter] Input event #3799        TOUCH 0 +4.0+1.0        key count: 0
[init -> event_filter] Input event #3800        TOUCH 0 +4.0+0.0        key count: 0
[init -> event_filter] Input event #3801        TOUCH 0 +3.0+0.0        key count: 0
[init -> event_filter] Input event #3802        TOUCH_RELEASE 0 key count: 0
```

Touch in the center area of the screen
```
[init -> event_filter] Input event #3803        TOUCH 0 +1764.0+1576.0  key count: 0
[init -> event_filter] Input event #3804        TOUCH 0 +1764.0+1577.0  key count: 0
[init -> event_filter] Input event #3805        TOUCH 0 +1765.0+1581.0  key count: 0
[init -> event_filter] Input event #3806        TOUCH 0 +1765.0+1580.0  key count: 0
[init -> event_filter] Input event #3807        TOUCH 0 +1766.0+1580.0  key count: 0
[init -> event_filter] Input event #3808        TOUCH_RELEASE 0 key count: 0
```

The more the touch is distant from the upper left corner the more it is offset.

> I could try reproducing this on the StarLite tablet if the scenarios are sufficiently similar.

As I mentioned I am not sure what to expect but judging by the touch coordinates, which are outside of the screen (the Fuji5b has a resolution of *1920x1080*) it appears they are not normalized/calibrated out-of-the-box.

jschlatow, 2025-04-07:
> As I mentioned I am not sure what to expect but judging by the touch coordinates, which are outside of the screen (the Fuji5b has a resolution of _1920x1080_) it appears they are not normalized/calibrated out-of-the-box.

Yes, it apparently needs normalization. You can use the `<transform>` filter for that. The scaling factor seems a bit odd, though.

jschlatow, 2025-04-07:
> If I move the mouse pointer via the touchscreen over any `menu_view` item in the runtime graph, e.g. `intel_fb`, the button is not outlined (thick black border). As soon as afterwards I move an external USB mouse device the border appears.

I'm sorry, it is still not 100% clear to me.
So, you're moving the mouse pointer by touching on a button in the runtime graph. What happens is a) nothing b) the button "interacts" but the black border is missing when the pointer still hovers over the button?

chelmuth, 2025-04-07:
> We identified commit [146f161](https://github.com/genodelabs/genode/commit/146f161383d8d488dee44279d370ad18d8676427) (gui_fader) as the culprit. Investigating...

Commit fb1096b466498bed2280794e9dc7c6f7632a0098 addresses the issue.

nfeske, 2025-04-07:
> On Sculpt 24.10 the button gets a black border when the mouse pointer is above the button, on 25.04 it does not.

While touched, the button should indeed show the hovered appearance.

jschlatow, 2025-04-08:
@cnuke I'm trying to build Sculpt 25.04 based on your branch _sculpt-25.04-staging-2025-04-04_. Unfortunately, I'm hitting an internal compiler error when building pkg/pc_wifi. Does this ring a bell?

```
genode$ ./tool/depot/create FORCE=1 UPDATE_VERSIONS=1 REBUILD=1 jschlatow_test/pkg/x86_64/pc_wifi
[...]
during RTL pass: reload
[...]/genode/depot/jschlatow_test/src/pc_wifi/2025-04-03/src/lib/wifi/socket_call.cc: In member function ‘void Lx::Socket::_do_poll_all()’:
[...]/genode/depot/jschlatow_test/src/pc_wifi/2025-04-03/src/lib/wifi/socket_call.cc:422:17: internal compiler error: Segmentation fault
  422 |                 }
      |                 ^
0x1baaff6 internal_error(char const*, ...)
        ???:0
0xcb96f1 lra_eliminate_regs_1(rtx_insn*, rtx_def*, machine_mode, bool, bool, poly_int<1u, long>, bool)
        ???:0
0xcb9f30 eliminate_regs_in_insn(rtx_insn*, bool, bool, poly_int<1u, long>)
        ???:0
0xcba679 lra_eliminate(bool, bool)
        ???:0
0xca1e35 lra(_IO_FILE*)
        ???:0
```

cnuke, 2025-04-08:
> @cnuke I'm trying to build Sculpt 25.04 based on your branch _sculpt-25.04-staging-2025-04-04_. Unfortunately, I'm hitting an internal compiler error when building pkg/pc_wifi. Does this ring a bell?

No, I had no issues building the pkg - I used Ubuntu 22.04.5 LTS with the normal Genode toolchain on one of our build-machines.

chelmuth, 2025-04-08:
> @cnuke I'm trying to build Sculpt 25.04 based on your branch _sculpt-25.04-staging-2025-04-04_. Unfortunately, I'm hitting an internal compiler error when building pkg/pc_wifi. Does this ring a bell?

Maybe https://github.com/genodelabs/genode/issues/5153 from March 2024? @cproc?

jschlatow, 2025-04-09:
@chelmuth Thanks for testing. Today, i.e. after a reboot, the issue is just gone. Yesterday, I tried several times on different branches and with a clean depot.

jschlatow, 2025-04-09:
> [@cnuke](https://github.com/cnuke) Thanks for clarification. Yes, touch clicks should actually work on 25.04. I'll try to reproduce this on the StarLite on which I had touch interaction in a usable state for 24.10.

I was able to reproduce the ineffective touch clicks on 25.04. @nfeske I'm afraid 3600705a2 broke the forwarding of touch and seq events to menu_view. I applied the following patch to nitpicker in order to restore the touch clicks on 25.04. I'm not sure, however, whether this contradicts the intention of 3600705a2.

``` diff
--- a/repos/os/src/server/nitpicker/user_state.cc
+++ b/repos/os/src/server/nitpicker/user_state.cc
@@ -277,8 +277,7 @@ void User_state::_handle_input_event(Input::Event ev)
                                return permitted ? _hovered : nullptr;
                        };

-                       if (ev.absolute_motion())
-                               receiver = abs_motion_receiver();
+                       receiver = abs_motion_receiver();
                }

                /*
```

nfeske, 2025-04-10:
I'm posting this from Sculpt OS 25.04 built with the current staging branch running on my Gen12 Framework laptop. Not a single issue so far.

ssumpf, 2025-04-10:
I have tested glmark2 on Fuji2 (Skylake) and Fuji5 (Tigerlake). Both machines hang (meaning execution freezes, no errors are reported, mouse & keyboard do not work any longer) reproducibly on current staging. Fuji2 hangs on the 2nd Linaro test and Fuji5 at the first green cat. I will try to find the last working commit.

ssumpf, 2025-04-10:
Follow up: With IOMMUs disabled (`<config ignore_drhd="yes" />` for acpi and without the `iommu_intel` argument on Nova kernel command line) the test succeeds on both machines. @jschlatow: Do you have any ideas? Could this be related to the problems in issue #5516?

nfeske, 2025-04-10:
@ssumpf, the symptoms sound indeed similar to those I experienced. Can you double check that this change https://github.com/genodelabs/genode/commit/221d51b58492bc85ab99fa10846f10aec5a77234 is present on your staging version? The fix entered staging just yesterday in the late afternoon.

ssumpf, 2025-04-10:
@nfeske: I used  staging from today and https://github.com/genodelabs/genode/commit/221d51b58492bc85ab99fa10846f10aec5a77234 is present (double checked).

nfeske, 2025-04-10:
Then I would not draw a connection.

ssumpf, 2025-04-10:
I will investigate.

ssumpf, 2025-04-10:
@jschlatow: I was able to identify the commit - which entered staging 30.01.25 - that triggers glmark2 to kind of freeze the machine as "platform/pc: use queued invalidation interface" (3c5b88111c87508bd639c8e06c104ffc41acb668). Since the behavior can be observed on current master/staging it is sufficient to execute glmark2 in a modular-sculpt scenario. Because I observed it on Skylake it should also trigger on x260. Since I am not familiar with DMA-remapping units I would appreciate your help.

jschlatow, 2025-04-11:
@ssumpf Can you please give fac485d a try.

nfeske, 2025-04-11:
Today, after the master branch got updated yesterday, the corresponding sculpt packages got published by genodelabs. Now would be a good time for the following steps:
* @chelmuth could you publish a current version of the touchpad driver and update the corresponding launcher?
* @jschlatow could you publish a current version of your screenshot pkg and update the laucher for the PinePhone?
* @cproc could you publish the current falkon browser and update the Sculpt preset(s) (PC, PinePhone, MNT)?
* @jschlatow you could now go forward updating Goa with the current genodelabs depot versions.
* Once Goa has become ready of Sculpt 25.04, @jschlatow could you update the Goa testbed preset(s)?
* With Goa up do date, all of us Goa-project maintainers could update their respective depots, offering an index for Sculpt 25.04, and thereby giving us all the great conveniences (msr gui, system info, ...) we enjoy on Sculpt.

ssumpf, 2025-04-11:
> [@ssumpf](https://github.com/ssumpf) Can you please give [fac485d](https://github.com/genodelabs/genode/commit/fac485db9f961ddf54ddd6403fa3e3845bc0fb44) a try.

It works 👍

jschlatow, 2025-04-11:
@nfeske I'm currently testing goa_testbed on Sculpt 25.04. I'm using my unix_shell runtime as payload. Unfortunately, I'm hitting the following two lines for each File_system session requested by the runtime:

```
[runtime -> recall_fs] Warning: re-attempted File_system session request 2 times (args: ram_quota=8K, cap_quota=0)
[runtime -> recall_fs] Warning: re-attempted File_system session request 4 times (args: ram_quota=16K, cap_quota=0)
```

Goa_testbed logs the following errors:
```
[monitor -> unix_shell -> vfs] Error: File_system-session creation failed (label="home -> /", ram_quota=192K, cap_quota=12, writeable=1, tx_buf_size=131072)
[monitor -> unix_shell -> vfs] Error: failed to create VFS node:    <fs label="home -> /"/>
[monitor -> unix_shell -> vfs] Error: File_system-session creation failed (label="report -> /", ram_quota=192K, cap_quota=12, writeable=0, tx_buf_size=131072)
[monitor -> unix_shell -> vfs] Error: failed to create VFS node:  <fs label="report -> /" writeable="no"/>
[monitor -> unix_shell -> vfs] Error: File_system-session creation failed (label="config -> /", ram_quota=192K, cap_quota=12, writeable=1, tx_buf_size=131072)
[monitor -> unix_shell -> vfs] Error: failed to create VFS node:  <fs label="config -> /"/>
[monitor -> unix_shell -> vfs] Error: File_system-session creation failed (label="shared -> /", ram_quota=192K, cap_quota=12, writeable=1, tx_buf_size=131072)
[monitor -> unix_shell -> vfs] Error: failed to create VFS node:  <fs label="shared -> /"/>
[monitor -> unix_shell -> vfs] Error: File_system-session creation failed (label="dotfiles -> /", ram_quota=192K, cap_quota=12, writeable=0, tx_buf_size=131072)
[monitor -> unix_shell -> vfs] Error: failed to create VFS node: 
[monitor -> unix_shell -> vfs]                  <fs label="dotfiles -> /" writeable="no"/>
```

Interestingly, _/ram/recall/goa_testbed/monitor/unix_shell/{home,report,config,dotfiles,shared}_ exist even though the session creation failed. Do you have any idea what is going wrong here?

chelmuth, 2025-04-11:
> * [@chelmuth](https://github.com/chelmuth) could you publish a current version of the touchpad driver and update the corresponding launcher?

The driver is available from chelmuth/pkg/pc_i2c_hid/2025-04-11 (commit 3b28ea3e5981de6ade8e8bb754216b2deb63366c).

jschlatow, 2025-04-11:
> [chocolate-doom] Error: In /home/jws/goarc:
>  Path variable 'versions_from_genode_dir' set to '/home/jws/develop/genode'
>  defines an invalid path. Valid paths are:
> 
>  /home/jws/develop/goa-projects/chocolate-doom
> 
>  You may consider setting '/home/jws/develop/goa-projects/chocolate-doom' in your $HOME/goarc or /goarc file.

Oh, the last line should actually be
> You may consider setting 'allowed_paths' in your $HOME/goarc or /goarc file.

Goa now maintains a list of allowed paths against which every path config variable is checked. By default, only the working directory is a valid path. You therefore need to explicitly add your genode dir to the allowed paths. See `goa help config` for details.

jschlatow, 2025-04-11:
screenshot/2025-04-11, goa_testbed/2025-04-11 and system_info/2025-04-11 are now available on my depot. Commit c94f480 updates the goa_testbed preset.

jschlatow, 2025-04-14:
> [@jschlatow](https://github.com/jschlatow): Could it be that [@nfeske](https://github.com/nfeske)'s screenshot pkg also started pkg/screenshot_trigger which would require Event and Gui? Because I am not able to take a screenshot with your pkg and screenshot_trigger seems to be the only one generating `KEY_PRINT` events.

@ssumpf I had looked at nfeske/pkg/screenshot/2024-10-09, which does not contain screenshot_trigger. Now I also had a look into nfeske/pkg/screenshot/2023-03-23, which indeed contains screenshot_trigger. Would you mind taking the latter as a blueprint for testing. If we settle on a runtime that works for the PinePhone, I can add this to my Goa project and publish it as a separate pkg (e.g. as screenshot_with_trigger) in the future.

nfeske, 2025-04-14:
> Interestingly, /ram/recall/goa_testbed/monitor/unix_shell/{home,report,config,dotfiles,shared} exist even though the session creation failed. Do you have any idea what is going wrong here?

@jschlatow, the directories are created by the chroot using its own file-system session. I have no explanation of the other observations. I can only guess that some adaptation to the root-dir-as-last-session-label-element change is missing somwhere along the chain of involved components.

alex-ab, 2025-04-14:
With commit 53fc636ee78654e6c0818df1920f95d5c072c374 and 0b61e88d9b3c962fc222306505bb56bde01cb65c the vesa_fb can be configured interactively via the Sculpt UI, tested in Qemu.

nfeske, 2025-04-14:
> [...] the vesa_fb can be configured interactively via the Sculpt UI, tested in Qemu.

@alex-ab works perfectly when running Sculpt in Qemu! Really cool. Immediately merged to staging.

jschlatow, 2025-04-15:
@alex-ab I tried this on the x230. The connectors report is available and I'm able to manually change the fb config to 1366x768. Yet, no UI appears when I click the vesa_fb.

cproc, 2025-04-15:
Commit 924e4a1 updates the falkon preset.

jschlatow, 2025-04-15:
> [@alex-ab](https://github.com/alex-ab) I tried this on the x230. The connectors report is available and I'm able to manually change the fb config to 1366x768. Yet, no UI appears when I click the vesa_fb.

When forcing it into using vesa_fb via `intel_gpu="no"`, the Sculpt manager still creates a Rom_handler for _/report/runtime/intel_fb/connectors_ before loading its config and falling back to vesa_fb. Yet, the Rom_handler is never reconstructed. This is fixed by ff33ceb.

jschlatow, 2025-04-16:
> [@jschlatow](https://github.com/jschlatow), the directories are created by the chroot using its own file-system session. I have no explanation of the other observations. I can only guess that some [...]

Turns out this was caused by a too long session label. Since chroot almost doubles the length of the session label, it quickly exceeds the maximum session-argument length. For the Sculpt release, I mitigated this by rewriting the identity in the goa_testbed preset (commit a73799c). I also created a separate issue #5527 to discuss adaptations to chroot.

alex-ab, 2025-04-16:
> When forcing it into using vesa_fb via `intel_gpu="no"`, the Sculpt manager still creates a Rom_handler for _/report/runtime/intel_fb/connectors_ before loading its config and falling back to [...]

@chelmuth, @nfeske: please consider for staging

nfeske, 2025-04-28:
@jschlatow I just tried using the screenshot utility from your index on my Sculpt PC and got the following error (I routed the target fs to ram_fs):
```
[runtime] [34mWarning: screenshot: no route to service "File_system" (label="screenshot -> target")
[runtime -> screenshot -> screenshot] Error: File_system-session creation failed (label="/", ram_quota=192K, cap_quota=12, writeable=1, tx_buf_size=131072)
[runtime -> screenshot -> screenshot] Error: failed to create VFS node: 
```

nfeske, 2025-04-28:
@jschlatow thanks a lot! Works as advertised now, once I manually added the capture policy in the nitpicker config. :-)

chelmuth, 2025-04-29:
Sculpt OS 25.04 was released with the changes in this issue.

closed 2025-04-29 by chelmuth