genodelabs/genode #5502: Refine `Ram_allocator` interface

nfeske, 2025-04-01:
The `Ram_allocator` is the lowest-level interface of the Genode API to allocate chunks of RAM backing store. It is typically used as in ingredient to higher-level allocators (like `Heap`), or for allocating shared-memory buffers (via `Attached_ram_dataspace`).

In its current form, it has the following limitations:
* For the common case of allocating RAM that can actually be accessed by the allocating component, the `Ram_allocator` must always be combined with a local `Region_map`. The higher-level allocators need to interact with both interfaces and handle their respective errors. The users of higher-level allocators need to follow the ceremony of passing `ram` and `rm` as sister arguments.
* The `Ram_allocator` talks about dataspaces. Within core, allocated RAM that is only used internally but never shared with other components still caries the cost of dataspaces. In particular, each dataspace consumes a capability.

These deficiencies can be addressed by changing the semantics of the `Ram_allocator`, speaking of locally mapped RAM blocks instead of dataspace capabilities. Each allocation would return a `Ram` object that holds the byte range of the local mapping and the dataspace capability (which may be invalid). The `Ram` object, holding a pointer, should not be copyable. It could also store a reference to the RAM allocator to automatically free the backing store at destruction time.

In the rare case where a RAM block should be allocated but not mapped - I'm thinking of DMA buffers allocated by the platform driver - the `Ram_allocator` interface may offer an `alloc_unmapped` operation. Alternatively, this (rare) scenario could be realized by operating on the PD session to manually allocate RAM dataspaces.

nfeske, 2025-04-06:
Commit https://github.com/genodelabs/genode/commit/c99c3a1abff5ddd62fdf51a1d72c40372dcaf60e introduces an exception-free paradigm for building allocators. The key element is the result type of allocation functions, which combines the `Attempt` approach (allocations can fail for allocator-specific reasons) with a unique-pointer that binds binds the lifetime of the allocation its own lifetime while also capturing the meta data of the individual allocation.

Commit https://github.com/genodelabs/genode/commit/a8832323bc3127940b31e7b3bb50bba375db3b2a applies the new approach to the RAM allocator.

Commit https://github.com/genodelabs/genode/commit/39c748cc7de6338a50c896c38660c396645c1350 applies it to the (byte-wise) allocators `Allocator`, `Range_allocator`, `Heap`, etc.

The new facilities challenge the existing notion of `Allocator` and `Range_allocator` as a common base class for all kinds of allocators, memory or not. E.g. in  5761148561da352bcd440593530d6b8a6706b4c3, I have applied the `Allocation` interface  to the allocation of trace-control slots. By moving into this direction, we not only attain the removal of exceptions but can also model allocator types strictly after their respective purpose.

nfeske, 2025-04-10:
This batch of changes have entered the master branch today. Subsequent steps will be discussed in separate issues.

closed 2025-04-10 by nfeske

alex-ab, 2025-11-07:
@chelmuth, @nfeske: please consider e994c57f4900b685745bed5b30b9a4179254b4f3 for staging. It fixes Sculpt boot issues. If more than one driver is used in a scenario, the MSI allocation will only work for the first, all other driver will be inoperable.