genodelabs/genode #5516: Replace use of local `Region_map` by dedicated `Local_rm`

nfeske, 2025-04-07:
The the decoupling of `Local_rm` from `Region_map` narrows the API, makes API-using code mode self-explanatory, and reduces the likelihood for API misuse (e.g., using a non-local `Region_map` for a `Heap`). The new type(s) also gives us a chance to clearly distinguish the `Env::Local_rm` (used by regular components) and `Core::Local_rm`. The latter does no longer need to be a `Region_map`, lifting complexity from core.

As another welcome opportunity, the abstraction for `Region_map` allows us to use the new `Allocation` utility for managing address-space attachments.

nfeske, 2025-04-08:
@cproc may you take care of the adaptation of Qt5 and Qt6 to this change?

nfeske, 2025-04-08:
With the current staging branch (including this change) I encounter a problem on my x260 with Sculpt, which I struggle to pin down. The problem occurs with base-hw and base-nova. The system either freezed in the middle the boot or behaves sluggish. Maybe @jschlatow has an idea?

The `while (!_empty())` loop at the end of `Intel::Queued_invalidator::invalidate_context` never completes. The head and tail pointers as read by `_empty` are running all the time. I guess that IOMMU faults are happening all the time? The busy-ness of the while loop explains the sluggyness, but what could be the reason for the infinite faults? When commenting out the construction the `_qeued_invalidator`, the system comes up on hw (but not on nova).

I suspect that some construction or population of an I/O-page table is automatically rolled back due to a missing `Allocator::deallocate = false`. But so far, I'm unable to spot it. @jschlatow, may you come to rescue?

jschlatow, 2025-04-09:
> With the current staging branch (including this change) I encounter a problem on my x260 with Sculpt, which I struggle to pin down. The problem occurs with base-hw and base-nova. The system either [...]

Not yet, but I'll look into that. Is this problem only observable on the x260?
 
> The `while (!_empty())` loop at the end of `Intel::Queued_invalidator::invalidate_context` never completes. The head and tail pointers as read by `_empty` are running all the time. I guess that [...]

Invalidation is not triggered by IOMMU faults but only by modifications to the I/O-page-table structures (e.g. when a Platform client allocates a DMA buffer or acquires a device). Actually, hardware should never touch the tail pointer. This is only done by software when a new invalidation command is inserted into the queue. Software then waits for the head pointer to catch up (done by hardware). So, I'm puzzled that you see the tail pointer moving at all.

nfeske, 2025-04-09:
@cproc thank for very much!

nfeske, 2025-04-09:
@jschlatow the queued-invalidation issue is a symptom of a cascade effect of a higher-level issue but is a still a bug that needs addressing. In `alloc_dma_buffer` there is guard that calls `domain.remove_range` for the presumably added range. However, in the concrete case, `domain.add_range` did not complete but triggered an exception during the page-table allocation. The page table is half-way constructed. When the ~Guard now calls domain.remove_range, the page-table entries for the range are only half-way constructed. But under this condition, the io_mmu.remove_range still issues the iotlb invalidation for the whole range. In this case, the iotlb invalidation is not needed because no DMA could have happened at this point. But when issued under this conditions, the queued invalidation seems to get stuck.

BTW, I could not again observe the running head and tail pointers. With the situation described above, neither pointer moves. 

I have conserved the current state along with my crude debug instrumentation on my `queued_invalidate` branch. @jschlatow I'd appreciate you having a look at this scenario because this can in principle happen rarely but surely. In my case, I just amplified the chance for triggering it. I'll concentrate now on the effect higher up the cascade. ;-)

jschlatow, 2025-04-09:
@nfeske Thanks for the detailed investigation. I'll look into that.

jschlatow, 2025-04-09:
> [@nfeske](https://github.com/nfeske) Thanks for the detailed investigation. I'll look into that.

I narrowed this down to an unhandled invalidation-queue error. Unfortunately, the x260 does not provide additional details on the cause of the error. I am going to work on implementing a proper recovery mechanism first and then see if other devices provide a hint on what is going wrong.

nfeske, 2025-04-09:
BTW, I pinned down the higher-level problem in 221d51b58492bc85ab99fa10846f10aec5a77234. With this patch (added to staging), Sculpt runs happily again. 

Just to explain what happened: Because of the accounting issue, the client (in this case the ahci driver) would repeatedly try to `alloc_dma_buffer`. Once the exception triggers (because the RAM quota depleted) during `add_range`, the `Guard` correctly did its job, reverting the RAM allocation. But the `Session_object::_ram_quota_guard` observed 0 as allocation size and could therefore not restore the original "used" amount. The client responded to the reflected OUT_OF_RAM by upgrading the session quota by 4K and trying the allocation again. This ended up in a loop because the upgraded quota is swallowed by the `_ram_quota_guard` right away.

To artificially create the invalidation issue with current staging, you may use b0bafb45b7ed8ee303f18ce5691851e2bdaec714 on my `trigger_iommu_invalidation_issue` branch I just pushed. It forcibly re-introduces the accounting issue.

closed 2025-04-10 by nfeske

jschlatow, 2025-04-11:
I created a separate issue #5523 for fixing the platform driver.

commit f7ed3246 (2025-04-08): base: introduce Local_rm for local Region_map

The new 'Local_rm' type offers a narrow interface for the interaction
with the component-local address space, managing the lifetime of
attachments by using the 'Allocation' API.

Fixes #5516