genodelabs/genode #5517: Rename `Xml_node`, `Xml_attribute`, and `Xml_generator` to `Node`, `Attribute`, and `Generator`

nfeske, 2025-04-08:
In anticipation of the potential change of Genode's universal use of XML to [another Syntax](http://genodians.org/nfeske/2024-12-20-moving-on-from-xml) and to improve the conciseness of our code, I'd like to drop the ceremonial `Xml_` prefix. We are inside the `Genode::` namespace after all. In the past 15 years, no other incarnations of `Node`, `Attribute`, and `Generator` tried to settle in this namespace. So we shouldn`t hesitate to make those terms first-level citizens of our API terminology.

nfeske, 2025-06-24:
The commit series introduces the syntax-agnostic `Node` API at base/node.h and converts init (and the sandbox library) to the new API.

For the most part, the conversion comes down to renaming `Xml_node` to `Node`. The helper utilities util/list_model.h and `with_matching_policy` have become compatible with both `Xml_node` and `Node`. The `from_xml` class functions for `Rect`, `Area`, `Point`, `Affinity` are now accompanied with `from_node` counterparts. This way, we can gradually update the code base.

nfeske, 2025-07-09:
Commit https://github.com/genodelabs/genode/commit/6ef6785e0defc835b983c546bc6a3138a01c1906 adds the syntax-agnostic `Generator` and `Generated_node` to base/node.h and introduces a `Dynamic_rom_session::Producer` for the use of The `Generator` interface.

Commit https://github.com/genodelabs/genode/commit/6ef6785e0defc835b983c546bc6a3138a01c1906 applies the use of `Generator` throughout Genode's code base.

cproc, 2025-07-11:
Commit 3ef0ba0 adapts Qt5 and Qt6 to the Generator change.

What would be the best way to use `Node` for the following part of the clipboard handling? `Node::content_size()` does not seem to be available.

https://github.com/cproc/qt6_base/blob/e1a4c2da5e9cdfa41f5a35316cc114efa8d7522d/src/plugins/platforms/genode/qgenodeclipboard.cpp#L96-L122

cproc, 2025-07-14:
Thanks, commit 2fc4dc6 adapts the Qt clipboard handling code accordingly.

ssumpf, 2026-01-19:
@nfeske: [682229c](https://github.com/ssumpf/genode/commit/682229c73a8ada4626ce0c2b5880b0ce81978450) uses `node` for the basic IP stack tests, please consider for staging.