genodelabs/genode #5519: tool/run/image/uboot: architecture may be wrong for FIT images

skalk, 2025-04-09:
When invoking mkimage with '-f auto' the tool uses heuristics to decide what kind of image to produce. It seems to be related to the version of mkimage and/or its input data, whether this decision is done right. In case of i.MX8 MQ EVK my mkimage tool (2023.01 on Debian 12.10) detects PowerPC as architecture :-(. Hinting the tool with the correct target architecture helps in my case.

nfeske, 2025-04-10:
Merged to staging.

chelmuth, 2025-04-11:
@a-dmg commit f9110541c834ad63576e2031897230adb9245db8 partially reverts your changes in 7d60b1f5e7dc3623651e76b56befa42dbcf0e981 (issue #5482). Do you see any conflicts with your use case?

a-dmg, 2025-04-11:
@chelmuth thank you for the heads up. This should not conflicts with our needs, indeed I overlooked the architecture, I should have left it. The branch we are trying to stay away is this one, introduced by this commit https://github.com/u-boot/u-boot/commit/bb07cdb19194bd141f7c3c631eddd4bfaf651e4d:
```c
if (IS_ENABLED(CONFIG_CMD_BOOTI) && images->os.arch == IH_ARCH_ARM64 && images->os.os == IH_OS_LINUX) {
...
}
```

Also, concerning the mkimage command, I have the intention to get rid of the dummy.dtb but wasn't successful yet.

commit f9110541 (2025-04-10): run/image/uboot: hint architecture for FIT images

When invoking mkimage with '-f auto' the tool uses heuristics to
decide what kind of image to produce. It seems to be related to
the version of mkimage and/or its input data, whether this decision
is done right. Therefore, clearly state the target architecture.

Fix #5519