genodelabs/genode #5525: depot: bash hash mismatch

b0inbot, 2025-04-14:
I thought my hash mismatch issues would be fixed with 016d63703d08afd6c20b57f0acdfa2363e6be350 but I am still getting conflicts with bash. At the bottom is the output. I will add some investigations here:

1. My system is fedora, however I tested this in a debian podman container as well. 
2. Running the hashing cmd in depot/genodelabs/src/bash/incomplete replicates the faebac... hash.
3. libc and ncurses do work, which is part of the commit linked above
4. Tried on master, staging, and various year-old tags, cleaning out depot/ and contrib/ each time. 
5. Tweaked locale settings around, no change in hash. 

```
penne[][genode] $ /home/boin/genode/genode/tool/ports/prepare_port bash
bash  extract bash-4.4.18.tar.gz (bash)
bash  apply src/noux-pkg/bash/build.patch
bash  apply src/noux-pkg/bash/check_dev_tty.patch
bash  apply src/noux-pkg/bash/disable_man2html.patch
bash  apply src/noux-pkg/bash/home_end_delete.patch
bash  apply src/noux-pkg/bash/bashrc.patch
bash  generate bash.hash

penne[000][genode] $ /home/boin/genode/genode/tool/depot/extract genodelabs/bin/x86_64/bash
created genodelabs/api/libc/2025-04-09
created genodelabs/api/noux/2023-06-15
created genodelabs/api/posix/2020-05-17

Error: /home/boin/genode/genode/repos/ports/recipes/src/bash/hash is out of date.

This error indicates that the archived source code has changed, which should
be reflected by incrementing the archive version and updating the hash of
the recipe. You may update the recipe hash via the following command:

  echo 2025-04-13 faebacf8bdca2ca8b82a2dffc607b4e6630df67e > /home/boin/genode/genode/repos/ports/recipes/src/bash/hash
```

chelmuth, 2025-04-14:
This is caused by some generated files that are part of the *port* and may differ between host systems. For testing I downloaded https://depot.genode.org/genodelabs/src/bash/2025-04-09.tar.xz, generated a local version via `./tool/depot/create genodelabs/src/bash FORCE=1 UPDATE_VERSIONS=1`, and compared both src-archive directory trees.

```
> diff -qr depot/genodelabs/src/bash/2025-04-14 /tmp/2025-04-09
Files depot/genodelabs/src/bash/2025-04-14/src/noux-pkg/bash/po/af.gmo and /tmp/2025-04-09/src/noux-pkg/bash/po/af.gmo differ
Files depot/genodelabs/src/bash/2025-04-14/src/noux-pkg/bash/po/bg.gmo and /tmp/2025-04-09/src/noux-pkg/bash/po/bg.gmo differ
Files depot/genodelabs/src/bash/2025-04-14/src/noux-pkg/bash/po/bg.po and /tmp/2025-04-09/src/noux-pkg/bash/po/bg.po differ
Files depot/genodelabs/src/bash/2025-04-14/src/noux-pkg/bash/po/ca.gmo and /tmp/2025-04-09/src/noux-pkg/bash/po/ca.gmo differ
Files depot/genodelabs/src/bash/2025-04-14/src/noux-pkg/bash/po/ca.po and /tmp/2025-04-09/src/noux-pkg/bash/po/ca.po differ
...
Files depot/genodelabs/src/bash/2025-04-14/src/noux-pkg/bash/po/zh_TW.po and /tmp/2025-04-09/src/noux-pkg/bash/po/zh_TW.po differ
```

b0inbot, 2025-04-15:
Interesting! Thank you

Those .gmo files do definitely change but those .po files... I would expect them to stay the same, unless the genode port is receiving specific localization changes that aren't being tracked as patches in the port folder?

Since it sounds like a non-issue, feel free to close or respond or do both.

chelmuth, 2025-04-29:
Close this issue as we do not plan to address the symptom currently.

closed 2025-04-29 by chelmuth