genodelabs/genode #5526: libc: add dummies for umtp(x) handling

atopia, 2025-04-14:
Add getutxent(), setutxent() and endutxent() dummy implementations.

chelmuth, 2025-04-29:
Please fix all spelling errors of *utmp*.

atopia, 2025-04-29:
Force-pushed commit [bdaa6b8](https://github.com/atopia/genode/commit/bdaa6b84949ca68fccb42e8411f1a3146a378c45) fixes the spelling.

skalk, 2025-04-30:
@atopia the changes of the libc most likely lead to the package build errors of coreutils and coreutils-minimal. Please check the mails of the nightly package buillds for most architectures.

atopia, 2025-04-30:
Sorry for the late reaction, [2b1c3e7](https://github.com/atopia/genode/commit/2b1c3e7033e5a6cb4bbd7ce1a4245624bdfaaf95) should fix the build problems.

nfeske, 2025-04-30:
Thanks. Merged to staging.

commit df720df2 (2025-04-29): libc: add dummies for utmp(x) handling

Add getutxent(), setutxent() and endutxent() dummy implementations.

Fixes #5526