genodelabs/genode #5529: base-hw: `Kernel panic on CPU 3`

trimpim, 2025-04-17:
In one of the tests in our continuous integration we have fairly regular appearing `Kernel panic on CPU 3`.

The test starts and stops the `fec_nic` driver on the imx8mp_iot_gate board.

The test is a bit sophisticated. The scenario hosts an SSH-server which writes the configuration for a `depot_query` / `depot_deploy` controlled sub-`init`. In this sub-`init` two different versions of the `fec_nic` are started alternately. After some iterations the Kernel panic occurs.

From the analysis @a-dmg and I did, it appears, that the fault occurs shortly after the new configuration for the sub-`init` is written, probably even before the new version of `fec_nic` is started (see attached logs). It is not even absolutely clear that the `fec_nic` causes the issue, but up until now we only did see the Kernel fault when this driver was restarted (we do restart other components / sub systems in our pipeline but `fec_nic` is restarted most often).

@a-dmg and I are speculating, that there might be a race condition, of an interrupt arriving while the old instance of the driver is stopped. We have a second imx8mp board (Phyboard-Pollux from Phytec) that shows the same errors.

[log_fail.txt](https://github.com/user-attachments/files/19793328/log_fail.txt)

[log_success.txt](https://github.com/user-attachments/files/19793331/log_success.txt)

[panic-1.txt](https://github.com/user-attachments/files/19793334/panic-1.txt)

[panic-2.txt](https://github.com/user-attachments/files/19793335/panic-2.txt)

[panic-3.txt](https://github.com/user-attachments/files/19793338/panic-3.txt)

chelmuth, 2025-04-17:
@trimpim @a-dmg do you entirely rule out memory corruption due to illegal DMA transactions of the FEC *after* the test mimic forcefully stops the driver? How do you ensure that the DMA engine in FEC does no longer process incoming packets and corrupts reused RAM regions?

In previous tests of a comparable scenario with stmmac NIC, you used `ping -p aa55 -s 1472` to stress the network with recognizable packet content, which produced the following log snippets.

```
[19158.087073]  Register dump:
[19158.087088]    x0    = 0xffffffd0157c40b0
[19158.109496]    x1    = 0x55aa55aa55aa55aa
[19158.109585]    x2    = 0x13f
```

```
[1102.902411]  Kernel: Unknown cpu exception EC=34 ISS=0 ip=0x55aa55aa55aa55aa
```

chelmuth, 2025-04-17:
I'd suggest to rule out potential DMA-based corruption first, as any analysis of directly or indirectly corrupted state is unrewarding in my eyes. @nfeske and me just had the idea to utilize the `managing_system` PD-session attribute to entirely isolate RAM allocations of the platform driver from the rest of the system. Optionally for testing, you may disable all RAM deallocations in core/kernel,thus that RAM is never reused. If the error does not occur then, you'd have a strong indicator that DMA is the cause.

skalk, 2025-04-17:
> I'd suggest to rule out potential DMA-based corruption first, as any analysis of directly or indirectly corrupted state is unrewarding in my eyes. [@nfeske](https://github.com/nfeske) and me just [...]

I would add: for testing and to rule whether DMA is the reason, you can in contrast to disable all RAM allocations in core/kernel do the same within the platform driver that delivers all DMA memory. Thereby, you save a bit memory in your dynamic scenario. Of course, you should spend the platform driver a lot more memory quota to let it do its job. I assume it would be enough to empty function `Session_component::_free_dma_buffer` in the platform driver.

If it turns out that DMA memory is not the reason here resp. in general: it would be helpful to use the provided/dumped information like instruction pointer and fault address to identify related source addresses using your concrete scenario's binary. From my perspective, I can only deduce very limited information from the register dump without the binary and its debug symbols.

skalk, 2025-05-05:
@trimpim that does not necessarily means that DMA errors are not an issue here, but it strongly indicates that there might be another issue as well.

To debug the kernel fault given the register dump, you especially need the core/kernel binary including its debug symbols. Therefore, please assure that you build core in your run-script, e.g. have a line like `build { core }` before `build_boot_image` gets called. Within you build directory, you'll find the core/kernel binary for debug purposes under `var/run/[run-script-name].core`. So in your case I would assume it is: `var/run/nic_driver_resilience.core`.
By using `genode-aarch-objdump -dlC var/run/nic_driver_resilience.core | less`, you can search for the address the instruction-pointer of the register-dump points to. It will hint you and us,which higher-level code raised the fault, and what concrete instructions. By sharing a fair scope of the objdump output together with the complete sources (e.g. correct branch), we can hopefully help you further. Alternatively, you can share the register-dump, the core/kernel binary with debug symbols, and the complete source branch (relating base/base-hw).

trimpim, 2025-05-06:
@skalk I created an archive containing
 - the complete log file
 - the `.core` file
 - the sources from `repos/base` and `repos/base-hw`

you can find the file at https://webmail2.trimpim.ch/gapfruit/trimpim/core_debug.tar.xz. If something is missing, I can update this file for you.

skalk, 2025-05-06:
@trimpim thanks for the bundle! I think I've a trace to the recent re-implementation of the paging across different cpu-cores in combination with schedule-"helping", see issue #5318. I'll investigate further, and provide a test-commit to you.

skalk, 2025-05-06:
@trimpim woould you please test commit 8f1c77a, whether it limits your nightly errors, and collect all other register-dumps that still occur?

trimpim, 2025-05-07:
@skalk I uploaded `v2` of the results. The error still occurred.

skalk, 2025-05-07:
> [@skalk](https://github.com/skalk) I uploaded `v2` of the results. The error still occurred.

Thank you @trimpim for the results. What a pity, so I'll re-investigate and try to provide a patch that at least reveals at which point an invalid Signal_context is called within the kernel.

skalk, 2025-05-07:
@trimpim I've prepared another commit to investigate first, which call of Signal_context::submit raised the fault. Please ignore the first commit, and replace it with f0d5c54.

trimpim, 2025-05-08:
@skalk I uploaded `v3` of the errors.

skalk, 2025-05-08:
@trimpim thank you, my initial thinking was wrong. The last results show it is no recent regression, but points to interrupt signaling during destruction of the driver. It might take some time to build a solution.

trimpim, 2025-05-08:
@skalk thanks for the information.

trimpim, 2025-05-14:
@skalk As I was not able to apply your patches to our `25.02` based branch, I created a separate branch which contains all our commits on top of `skalk/merge_to_staging`.  For this I do have build the new tool chain first.

Will I have to get some special branch for the `genode-imx` / `gneode-world` repository? or can I just pick `genodelabs/staging` of the two?

trimpim, 2025-05-14:
@skalk when I do `make -C build/arm_v8a KERNEL=hw BOARD=imx8mp_iot_gate run/log` to build 1dc89a547435b6031ec5247e787c4a0aeb9382fe with the newly built toolchain I get the following error:

```
    COMPILE  kernel/thread.o
make[3]: *** No rule to make target '/data/genode/repos/base-hw/src/core/kernel/vm_thread_on.cc', needed by 'kernel/vm_thread_on.o'.  Stop.
make[3]: *** Waiting for unfinished jobs....
    COMPILE  kernel/timer.o
make[2]: *** [var/libdeps:213: core-hw-imx8mp_iot_gate.lib.a] Error 2
make[1]: *** [Makefile:351: gen_deps_and_build_targets] Error 2
make[1]: Leaving directory '/data/genode/build/arm_v8a'
Error: Genode build failed
```

atopia, 2025-05-14:
Without trying myself, such errors are usually caused by an old object file in the build directory that does not have a corresponding source file in the new source tree. Should be fixed by a
```
make -C build/arm_v8a clean
```

trimpim, 2025-05-14:
@atopia  thanks for the input, unfortunately this didn't help in this case.

atopia, 2025-05-14:
Sorry to hear that, I didn't mean to suggest something obvious that you have probably already tried. If memory serves right I might have manually removed the offending object file in the build directory when encountering similar problems (not really a solution to the problem, but a possible workaround).

skalk, 2025-05-14:
@trimpim that is unfortunate that it is not easy to backport the patch - there are indeed quite a lot things touched in the base-framework right now. But actually I cannot share your observation about a failing built for your designated target. I did not execute the outcome on that specific hardware target, but I can build it for imx8mp_iot_gate, and I've executed it successfully on another target. Either you use an outdated genode-imx branch, or you have build artefacts somewhere, like @atopia suggested. The compilation unit `vm_thread_on.cc` got renamed to vcpu_thread_on.cc` in sculpt release 25.04.

trimpim, 2025-05-14:
@skalk sorry for causing trouble, the problem was as you pointed already out an outdated version of `genode-imx`

trimpim, 2025-05-14:
@skalk I tried to build our scenario so I could run the tests.

Due to the changes to the `Root` interface this requires changes to all our servers. As I'm in vacation form 2025-05-16 to 2025-05-31,  I'm currently not able to test you changes.