genodelabs/genode #5530: Address invalid opcode exception in KVM

cnuke, 2025-04-23:
Running the Genodians appliance in KVM with the E1000e device-model produced a  *invalid opcode* exception because a `movd` instructions was used for memory-mapped I/O access.

cnuke, 2025-04-23:
The following commits address this problem by first preventing the generation of the potentially problematic instructions and second by making sure that the proper calls are being made where such optimization cannot take place.

* [085cbcb](https://github.com/cnuke/genode/commit/085cbcb33ecf84efc83ec4df3c50eeba9308e326) sets the common *-mno-* flags also set when normally building the kernel to prevent making use of certain features.
* [e7af1f6](https://github.com/cnuke/genode/commit/e7af1f61ce316b581b5eda98a1a765da56efbe39) extendeds our `asm/io.h` shadow header with the specific mmio accessor functions.

skalk, 2025-05-05:
@chelmuth @cnuke I'm fine with it

chelmuth, 2025-05-28:
Merged to 25.05.

closed 2025-05-28 by chelmuth