genodelabs/genode #5531: run/thread: test sometimes get stuck

rite, 2025-04-23:
When repeatedly executing `run/thread` with base-hw on a [PC Engines apu4](https://www.pcengines.ch/apu4d4.htm), the run script eventually times out (after several successful executions):

```
4066 MiB RAM and 64536 caps assigned to init
[init -> test-thread] --- thread test started ---
[init -> test-thread] destroy dependent CPU sessions in wrong order
[init -> test-thread] running 'test_stack_alloc'
```

I was able to reproduce this behavior with current genode/master (25.02-129-g016d63703d). I don't mind this problem in case it applies only to this particular hardware (or AMD-based x86). However I have no idea whether (or to what degree) it is hardware related.

chelmuth, 2025-04-29:
I ran `(for d in $(seq 100); do make -C build/x86_64 run/thread KERNEL=hw BOARD=pc || exit 1; done)` in Qemu on 3f905937025f64e781413048462bd6c175b9f3ca but did experience the hang. The DEBUG commit removes all test steps after `test_stack_alloc` to minimize runtime.

- Does this approach reflect your test pattern?
- Did you try to stress a different hardware / Qemu already?

rite, 2025-04-30:
> Does this approach reflect your test pattern?

Yes. I was able to reproduce the hang on the PC Engines by running `(for d in $(seq 100); do make -C build/x86_64 run/thread KERNEL=hw BOARD=pc || exit 1; done)` with 3f90593.

> Did you try to stress a different hardware / Qemu already?

I just did, using your branch: In Qemu, I also _didn't_ experience the hang neither with KVM enabled or not. However, the hang is reproducible on a UP Xtreme Edge.

alex-ab, 2025-05-05:
> I was able to reproduce this behavior with current genode/master (25.02-129-g016d63703d). I don't mind this problem in case it applies only to this particular hardware (or AMD-based x86). However I have no idea whether (or to what degree) it is hardware related.

Just as data point, because I stumbled by accident over it today: In our nightly test infrastructure on Intel hardware the test failed also, so it is not specific for AMD at all - I would guess. At that day, all other run/thread across all other platforms and kernels succeeded, so it is also not related to the current changes on staging. 
```
[2025-05-03 08:11:07] kernel initialized
[2025-05-03 08:11:07] Genode sculpt-25.04-12-gda44081b6c9
[2025-05-03 08:11:07] 16244 MiB RAM and 64536 caps assigned to init
[2025-05-03 08:11:07] [init -> test-thread] --- thread test started ---
[2025-05-03 08:11:07] [init -> test-thread] destroy dependent CPU sessions in wrong order
[2025-05-03 08:11:07] [init -> test-thread] running 'test_stack_alloc'
[2025-05-03 08:12:07] Error: Test execution timed out
```

chelmuth, 2025-05-05:
Interesting, a grep in the log files since 2024-01-01 revealed the following hits for timeout after *running 'test_stack_alloc'*.

```
[2024-07-02 08:47:42]
[2024-11-06 07:37:47]
[2024-12-16 07:36:45]
[2025-02-14 08:39:17]
[2025-02-18 08:16:53]
[2025-03-10 09:38:41]
[2025-03-31 05:52:21]
[2025-04-10 08:07:11]
[2025-04-16 08:17:40]
[2025-05-03 08:11:07]
```

chelmuth, 2025-05-28:
The 25.05 release still produces this error.

```
[2025-05-28 07:41:05] [init -> test-thread] running 'test_stack_alloc'
```