genodelabs/genode #5532: sculpt: KEY_ENTER propagates to Inspect and File view when entering wifi passphrase

jschlatow, 2025-04-24:
When entering the passphrase in Sculpt's network dialog, pressing KEY_ENTER can be used as an alternative to clicking the "connect" button. At the same time, KEY_ENTER is also propagated to the Inspect view or the File view (if enabled). This is due to the fact that `Keyboard_focus` already generates a focus report with `label="wm ->"` before the input event is submitted to the GUI client.

This is reproducible with Sculpt 24.10 and Sculpt 25.04.

jschlatow, 2025-04-24:
I do not consider this critical for the Sculpt release. @nfeske Unless you have a smart idea how to fix this, I'd rather address this after the release.

nfeske, 2025-04-27:
Thank you opening the issue. I'm afraid that there is no simple fix. I agree with deferring the issue to the time after the release.

nfeske, 2025-10-20:
The patch looks good to me. It is perfectly in line with the intuition that the focus cannot change while a key is held after all.

closed 2025-10-30 by chelmuth