genodelabs/genode #5540: tool chain 25.05 update

cproc, 2025-05-02:


cproc, 2025-05-02:
A first version of GCC 14.2.0 and binutils 2.44 can be built with my [issue5540](https://github.com/cproc/genode/tree/issue5540) branch.

Once the new compile errors when building Genode with this updated tool chain are fixed we can try to unify types like `uint64_t` with the Genode and libc types.

I'm planning to update GDB as well.

nfeske, 2025-05-07:
@cproc really cool to see all the GCC14 issues springing up and be solved! Looking at the delightful correspondence between @ssumpf and you put a smile on my face. :)

I wonder, would now be a good time to bring the existing commit series to staging? If you think so, I'd very much appreciate a branch rebased on staging that contains all the commits needed. (don't worry that not all issues are solved already - that's what the staging branch is for, after all) If it is too early, please give me a notice once you think the time is right.

cproc, 2025-05-07:
@nfeske: I rebased my issue5540 branch on staging. With the latest fixes of the compile error GitHub issues the state is good enough to build the Sculpt image for NOVA.

nfeske, 2025-05-07:
Super cool! Merged to staging right away.

skalk, 2025-05-07:
On our test-infrastructure Ubuntu 24.04 the toolchain script resp. compiling of binutils for x86 fails at the very beginning with:
```
/usr/bin/ld: /usr/bin/ld: /usr/bin/ld: /usr/bin/ld: ./.libs/libgprofng.a(DbeSession.o): in function `DbeSession::ql_parse(char const*)':
/home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3009:(.text+0x9d3b): undefined reference to `QL::Parser::Parser(QL::Result&)'
/usr/bin/ld: /home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3010:(.text+0x9d43): undefined reference to `QL::Parser::parse()'
/usr/bin/ld: /home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3013:(.text+0x9d7b): undefined reference to `QL::Parser::~Parser()'
/usr/bin/ld: /home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3013:(.text.unlikely+0x686): undefined reference to `QL::Parser::~Parser()'
./.libs/libgprofng.a(DbeSession.o): in function `DbeSession::ql_parse(char const*)':
/home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3009:(.text+0x9d3b): undefined reference to `QL::Parser::Parser(QL::Result&)'
/usr/bin/ld: /home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3010:(.text+0x9d43): undefined reference to `QL::Parser::parse()'
/usr/bin/ld: /home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3013:(.text+0x9d7b): undefined reference to `QL::Parser::~Parser()'
/usr/bin/ld: /usr/bin/ld: /home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3013:(.text.unlikely+0x686): undefined reference to `QL::Parser::~Parser()'
./.libs/libgprofng.a(DbeSession.o): in function `DbeSession::ql_parse(char const*)':
/home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3009:(.text+0x9d3b): undefined reference to `QL::Parser::Parser(QL::Result&)'
/usr/bin/ld: /home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3010:(.text+0x9d43): undefined reference to `QL::Parser::parse()'
/usr/bin/ld: /home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3013:(.text+0x9d7b): undefined reference to `QL::Parser::~Parser()'
/usr/bin/ld: /home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3013:(.text.unlikely+0x686): undefined reference to `QL::Parser::~Parser()'
./.libs/libgprofng.a(DbeSession.o): in function `DbeSession::ql_parse(char const*)':
/home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3009:(.text+0x9d3b): undefined reference to `QL::Parser::Parser(QL::Result&)'
/usr/bin/ld: /home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3010:(.text+0x9d43): undefined reference to `QL::Parser::parse()'
/usr/bin/ld: /home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3013:(.text+0x9d7b): undefined reference to `QL::Parser::~Parser()'
/usr/bin/ld: /home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3013:(.text.unlikely+0x686): undefined reference to `QL::Parser::~Parser()'
./.libs/libgprofng.a(DbeSession.o): in function `DbeSession::ql_parse(char const*)':
/home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3009:(.text+0x9d3b): undefined reference to `QL::Parser::Parser(QL::Result&)'
/usr/bin/ld: /home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3010:(.text+0x9d43): undefined reference to `QL::Parser::parse()'
/usr/bin/ld: /home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3013:(.text+0x9d7b): undefined reference to `QL::Parser::~Parser()'
/usr/bin/ld: /home/kalle/src/genode/contrib/binutils-35d7d34e522adc7f4374415b5932a5edee1ddcae/src/noux-pkg/binutils/gprofng/src/DbeSession.cc:3013:(.text.unlikely+0x686): undefined reference to `QL::Parser::~Parser()'
  CC       eelf_i386.o
collect2: error: ld returned 1 exit status
make[6]: *** [Makefile:695: gprofng] Error 1
make[6]: *** Waiting for unfinished jobs....
collect2: error: ld returned 1 exit status
make[6]: *** [Makefile:707: gprofng-display-src] Error 1
collect2: error: ld returned 1 exit status
collect2: error: ld returned 1 exit status
make[6]: *** [Makefile:699: gprofng-archive] Error 1
collect2: error: ld returned 1 exit status
make[6]: *** [Makefile:703: gprofng-collect-app] Error 1
make[6]: *** [Makefile:711: gprofng-display-text] Error 1
make[5]: *** [Makefile:572: all] Error 2
make[4]: *** [Makefile:472: all-recursive] Error 1
[... cut ...]
  CC       eelf_x86_64.o
  CCLD     ld-new
make[1]: *** [Makefile:1028: all] Error 2
make: *** [tool/tool_chain:387: /home/kalle/src/genode/build/tool_chain-25.05/bootstrap/binutils/binutils/objdump] Error 2
```
The reason is again: the typical problem of checked out sources with pre-built yacc/bison files, where the timestamp is altered in a way that it re-tries to build the grammar file, which fails. Touching of `/src/noux-pkg/binutils/gprofng/src/QLParser.tab.cc` helped me. Commit d211147 references this problem.

nfeske, 2025-05-07:
This reminds me very much of https://github.com/genodelabs/genode/blob/staging/tool/ports/mk/install.mk#L283-285.

@cproc may we try to explicitly prevent the execution of those steps (in the spirit of `_discharge_automake`) by patching the rules as an alternative to the tweaking of timestamps?

nfeske, 2025-05-07:
The `tool_chain` script worked nicely for me. :)

cproc, 2025-05-07:
> This reminds me very much of https://github.com/genodelabs/genode/blob/staging/tool/ports/mk/install.mk#L283-285.
>
> [@cproc](https://github.com/cproc) may we try to explicitly prevent the execution [...]

Commit f67e882 adds a discharge pattern for `*.yy` files.

nfeske, 2025-05-07:
> Commit https://github.com/genodelabs/genode/commit/f67e882d070b86336be28ae3e66d852d4aca0650 adds a discharge pattern for *.yy files.

Excellent! Merged to staging.

nfeske, 2025-05-08:
> Commit https://github.com/genodelabs/genode/commit/f67e882d070b86336be28ae3e66d852d4aca0650 adds a discharge pattern for *.yy files.

This seems to have an unwelcome effect on the bash port. :-/

nfeske, 2025-05-08:
The error message is:
```
      [bash]  /data/depot/checker/src/bash/2025-05-07/src/noux-pkg/bash/lib/intl/plural.c:146:10: fatal error: plural.h: No such file or directory
      [bash]    146 | #include "plural.h"
```

cproc, 2025-05-08:
> The error message is:
> 
> ```
>       [bash]  /data/depot/checker/src/bash/2025-05-07/src/noux-pkg/bash/lib/intl/plural.c:146:10: fatal error: plural.h: No such file or directory
>       [bash]    146 | #include "plural.h"
> ```

It seems to be a similar problem, but not related to the `*.yy` discharge pattern. Apparently `plural.c` sometimes gets regenerated from `plural.y`, but I can't find a corresponding rule in `Makefile.in`.

cproc, 2025-05-08:
> It seems to be a similar problem, but not related to the `*.yy` discharge pattern. Apparently `plural.c` sometimes gets regenerated from `plural.y`, but I can't find a corresponding rule in `Makefile.in`.

e2fsprogs has the same problem (issue #5535).

skalk, 2025-05-08:
> e2fsprogs has the same problem (issue [#5535](https://github.com/genodelabs/genode/issues/5535)).

I recently addressed that by issue #5535

nfeske, 2025-05-14:
@cproc I've reverted the "prepare_ports: prevent regeneration from '*.yy' files" commit for now because of the big impact of the missing build of bash on many tests. I guess the wildcard of the added discard pattern is too broad. So we need to figure out more precisely the difference between the tool chain port and the bash port. If we are unable to come up with a suitable pattern, we may fall back to @skalk's approach.

cproc, 2025-05-14:
@nfeske: did you see my previous comment about the bash issue?

nfeske, 2025-05-14:
@cproc I did but have possibly drawn the wrong conclusion. Having observed that the bash prepare error (presumably) started to appear with the discharge change, I went for the rollback - arguably a little bit desperate.

In any case, I'd very much appreciate your advice how to proceed. ;-)

chelmuth, 2025-05-14:
Regarding the .yy issue: I moved the port of bash to a tmp location and recreated it with staging as of yesterday on the CI server. The generated source trees differ (esp. in the file plural.c).

```
--- bash-74368ed8f741b5f9a2a20342f9d6dcf5e9228c38/src/noux-pkg/bash/lib/intl/plural.c   2025-05-14 13:38:56.022073766 +0200
+++ bash-74368ed8f741b5f9a2a20342f9d6dcf5e9228c38.old/src/noux-pkg/bash/lib/intl/plural.c       2025-05-08 06:04:39.209834952 +0200
@@ -1,31 +1,42 @@
-/* A Bison parser, made by GNU Bison 2.0.  */
+/* A Bison parser, made by GNU Bison 3.8.2.  */
```

I suggest to remove the revert commit again. @cproc do you agree? Are there more ports affected?

nfeske, 2025-05-14:
I just reverted the revert. Thanks @chelmuth for lifting the clouds.

cproc, 2025-05-14:
There is this rule in bash `Makefile.in`:

```
plural.lo: $(srcdir)/plural.c
    $(LIBTOOL) --mode=compile $(COMPILE) $(srcdir)/plural.c
```

but apparently no explicit rule which states that `plural.c` depends on `plural.y`.

Other ports with the same rule are `e2fsprogs`, `gcc` and `ltris`. `gdb` has a `plural.y` too, but with a different rule:

```
plural-config.h: plural.c
```

ssumpf, 2025-05-14:
@nfeske: [e9f3335](https://github.com/ssumpf/genode-world/commit/e9f33351aa2965acb831293ced217d38d08a32d6) fixes JDK.

nfeske, 2025-05-15:
@ssumpf thanks a lot! Merged to staging. BTW, I'm not against adding the `template` keyword at the base header to maintain C++17 compatibility for a bit longer. I've not deliberately made it incompatible.

chelmuth, 2025-05-15:
spark is fixed on staging

skalk, 2025-05-15:
Commit 4ad90f9 fixes the sdl_image compilation and package

skalk, 2025-05-15:
Commit c24b2d9 fixes libgetdns compile errors

nfeske, 2025-05-15:
Thanks @skalk. I merged both commits to staging.

chelmuth, 2025-05-15:
@cproc I currently have no clue what's different on arm and x86 in #5574.

atopia, 2025-05-15:
With the new toolchain I'm getting compile errors when using libstdc++ headers (`chrono` to be precise), as an example:
```
/home/ben/dev/3rd-genode/depot/genodelabs/api/stdcxx/2025-05-15/include/stdcxx/std/limits:1989:1: error: '_Float32' was not declared in this scope; did you mean '_Float16'?
 1989 | __glibcxx_float_n(32)
```
and, similarly
```
/home/ben/dev/3rd-genode/depot/genodelabs/api/stdcxx/2025-05-15/include/stdcxx/std/format:2243:38: error: template argument 1 is invalid
 2243 |     struct formatter<_Float64, _CharT>
```
The component is a very basic C++ helper program built via Goa that did compile with the old tool chain.

I have the hunch that this might be connected to #5567 and maybe the `floatn.h` header, but I haven't been able to figure out where the `_Float` definitions are supposed to come from.

atopia, 2025-05-15:
[b22f5f6](https://github.com/atopia/genode/commit/b22f5f637970193ec56e5c6f7d8407fe0b04986c) should fix building with newer python versions. Unfortunately I made a mistake in the initial version of the commit and I won't be able to build the tool chain locally until Monday morning. Feel free to merge, otherwise I will report back on Monday.

cproc, 2025-05-15:
> I have the hunch that this might be connected to [#5567](https://github.com/genodelabs/genode/issues/5567) and maybe the `floatn.h` header, but I haven't been able to figure out where the `_Float` definitions are supposed to come from.

They are supposed to be built-in with GCC 14. Could it be possible that the old tool chain is still used to build the file which includes the stdcxx 14 headers?

chelmuth, 2025-05-19:
Commit 3ed32af36e530f8099c2f220ba1c2170dadc455e addresses the following build errors on riscv.

```
In file included from /data/depot/checker/api/vfs/2025-05-18/include/vfs/readonly_value_file_system.h:19,
                 from /data/depot/checker/src/vfs_legacy_oss/2025-05-18/src/lib/vfs/legacy_oss/vfs_oss.cc:22:
/data/depot/checker/api/vfs/2025-05-18/include/vfs/readonly_value_file_system.h: In instantiation of ‘class Vfs::Readonly_value_file_system<unsigned int>’:
/data/depot/checker/src/vfs_legacy_oss/2025-05-18/src/lib/vfs/legacy_oss/vfs_oss.cc:152:26:   required from here
  152 |                                 _channels_fs         .value(channels);
      |                                                      ^
/data/depot/checker/api/vfs/2025-05-18/include/vfs/single_file_system.h:229:22: error: ‘virtual void Vfs::Single_file_system::close(Vfs::Vfs_handle*)’ was hidden [-Werror=overloaded-virtual=]
  229 |                 void close(Vfs_handle *handle) override
      |                      ^~~~~
/data/depot/checker/api/vfs/2025-05-18/include/vfs/readonly_value_file_system.h:170:22: note:   by ‘void Vfs::Readonly_value_file_system< <template-parameter-1-1>, BUF_SIZE>::close(Vfs::Vfs_watch_handle*) [with T = unsigned int; unsigned int BUF_SIZE = 128]’
  170 |                 void close(Vfs_watch_handle *handle) override
      |                      ^~~~~
```

I can't imagine why the warning is only produced on riscv and not on other architectures but according to the discussion [here](https://stackoverflow.com/questions/9995421/gcc-woverloaded-virtual-warnings) it is valid and can be addressed as in my commit. Opinions?

atopia, 2025-05-19:
> They are supposed to be built-in with GCC 14. Could it be possible that the old tool chain is still used to build the file which includes the stdcxx 14 headers?

Thanks for the hint, that was indeed the problem.

atopia, 2025-05-19:
> [b22f5f6](https://github.com/atopia/genode/commit/b22f5f637970193ec56e5c6f7d8407fe0b04986c) should fix building with newer python versions. Unfortunately I made a mistake in the initial version of [...]

With the fix, GDB builds successfully with Python 3.13.

chelmuth, 2025-05-19:
> Commit [3ed32af](https://github.com/genodelabs/genode/commit/3ed32af36e530f8099c2f220ba1c2170dadc455e) addresses the following build errors on riscv.
> 
> ```
> In file included from /data/depot/checker/api/vfs/2025-05-18/include/vfs/readonly_value_file_system.h:19,
>                  from /data/depot/checker/src/vfs_legacy_oss/2025-05-18/src/lib/vfs/legacy_oss/vfs_oss.cc:22:
> /data/depot/checker/api/vfs/2025-05-18/include/vfs/readonly_value_file_system.h: In instantiation of ‘class Vfs::Readonly_value_file_system<unsigned int>’:
> /data/depot/checker/src/vfs_legacy_oss/2025-05-18/src/lib/vfs/legacy_oss/vfs_oss.cc:152:26:   required from here
>   152 |                                 _channels_fs         .value(channels);
>       |                                                      ^
> /data/depot/checker/api/vfs/2025-05-18/include/vfs/single_file_system.h:229:22: error: ‘virtual void Vfs::Single_file_system::close(Vfs::Vfs_handle*)’ was hidden [-Werror=overloaded-virtual=]
>   229 |                 void close(Vfs_handle *handle) override
>       |                      ^~~~~
> /data/depot/checker/api/vfs/2025-05-18/include/vfs/readonly_value_file_system.h:170:22: note:   by ‘void Vfs::Readonly_value_file_system< <template-parameter-1-1>, BUF_SIZE>::close(Vfs::Vfs_watch_handle*) [with T = unsigned int; unsigned int BUF_SIZE = 128]’
>   170 |                 void close(Vfs_watch_handle *handle) override
>       |                      ^~~~~
> ```
> 
> I can't imagine why the warning is only produced on riscv and not on other architectures but according to the discussion [here](https://stackoverflow.com/questions/9995421/gcc-woverloaded-virtual-warnings) it is valid and can be addressed as in my commit. Opinions?

@nfeske what do you think? I did not merge the commit to staging yet.

nfeske, 2025-05-19:
@chelmuth I'm fine with it. Please go ahead.

cproc, 2025-05-19:
Commit 2dd3f64 updates GDB to version 16.3.

closed 2025-05-28 by chelmuth