genodelabs/genode #5543: text_area: compile error with GCC 14

cproc, 2025-05-02:
When building `app/text_area` with GCC 14, the following error occurs:

```
In file included from /.../repos/gems/include/dialog/runtime.h:17,
                 from /.../repos/gems/src/app/text_area/main.cc:17:
/.../repos/gems/include/dialog/sandboxed_runtime.h:270:30: error: 'struct Genode::Avl_node<Dialog::Sandboxed_runtime::View> Genode::Avl_node<Dialog::Sandboxed_runtime::View>::Avl_node' is private within this context
  270 |                 friend class Avl_node<View>;
      |                              ^~~~~~~~~~~~~~
In file included from /.../repos/gems/include/dialog/sandboxed_runtime.h:17:
/.../repos/base/include/util/dictionary.h:35:23: note: declared private here
   35 |                 class Element : private Avl_node<T>
      |                       ^~~~~~~
```

cproc, 2025-05-02:
Fixed by 0cc9070.

chelmuth, 2025-05-05:
While I do not understand why the patch helps in this context, I'm wondering if a `using namespace Genode` declaration in `namespace Dialog` could be a better fit.

cproc, 2025-05-05:
> While I do not understand why the patch helps in this context, I'm wondering if a `using namespace Genode` declaration in `namespace Dialog` could be a better fit.

My understanding is that the compiler matched `Avl_node<View>` with the private `Avl_node<T>` base class instead of `Genode::Avl_node<View>`. I tried `using namespace Genode` in `namespace Dialog` now, but it did not fix the error unfortunately.

commit bfd36f36 (2025-05-05): gems: fix compile error with GCC 14 in sandboxed_runtime.h

Fixes #5543
closed 2025-05-28 by chelmuth