genodelabs/genode #5549: ccid: compile errors with GCC 14

cproc, 2025-05-05:
When building `lib/ccid` with GCC 14, the following errors occur:

```
/.../contrib/ccid-7a9fb9a6d004391a83a7c6c827bdacbc785c9353/src/lib/ccid/src/towitoko/pps.c: In function 'PPS_Exchange':
/.../contrib/ccid-7a9fb9a6d004391a83a7c6c827bdacbc785c9353/src/lib/ccid/src/towitoko/pps.c:83:3: error: implicit declaration of function 'memcpy' [-Wimplicit-function-declaration]
   83 |   memcpy (params, confirm, len_confirm);
      |   ^~~~~~
/.../contrib/ccid-7a9fb9a6d004391a83a7c6c827bdacbc785c9353/src/lib/ccid/src/towitoko/pps.c:36:1: note: include '<string.h>' or provide a declaration of 'memcpy'
   35 | #include "debug.h"
  +++ |+#include <string.h>
   36 | 
/.../contrib/ccid-7a9fb9a6d004391a83a7c6c827bdacbc785c9353/src/lib/ccid/src/towitoko/pps.c:83:3: warning: incompatible implicit declaration of built-in function 'memcpy' [-Wbuiltin-declaration-mismatch]
   83 |   memcpy (params, confirm, len_confirm);
      |   ^~~~~~
/.../contrib/ccid-7a9fb9a6d004391a83a7c6c827bdacbc785c9353/src/lib/ccid/src/towitoko/pps.c:83:3: note: include '<string.h>' or provide a declaration of 'memcpy'
/.../contrib/ccid-7a9fb9a6d004391a83a7c6c827bdacbc785c9353/src/lib/ccid/src/towitoko/pps.c: In function 'PPS_Match':
/.../contrib/ccid-7a9fb9a6d004391a83a7c6c827bdacbc785c9353/src/lib/ccid/src/towitoko/pps.c:94:7: error: implicit declaration of function 'memcmp' [-Wimplicit-function-declaration]
   94 |       memcmp (request, confirm, len_request))   /* different contents */
      |       ^~~~~~
/.../contrib/ccid-7a9fb9a6d004391a83a7c6c827bdacbc785c9353/src/lib/ccid/src/towitoko/pps.c:94:7: note: include '<string.h>' or provide a declaration of 'memcmp'
/.../contrib/ccid-7a9fb9a6d004391a83a7c6c827bdacbc785c9353/src/lib/ccid/src/towitoko/pps.c:94:33: warning: 'memcmp' argument 3 type is 'unsigned int' where 'long unsigned int' is expected in a call to built-in function declared without prototype [-Wbuiltin-declaration-mismatch]
   94 |       memcmp (request, confirm, len_request))   /* different contents */
      |                                 ^~~~~~~~~~~
<built-in>: note: built-in 'memcmp' declared here
```

cproc, 2025-05-07:
Fixed by 4ada102.

nfeske, 2025-05-14:
Merged to staging.

closed 2025-05-28 by chelmuth