genodelabs/genode #5550: libdrm: compile error with GCC 14

cproc, 2025-05-05:
When building `lib/libdrm` with GCC 14, the following error occurs:

```
/.../contrib/libdrm-cacc86de9adf92f7dc7212928555d9e7f640da76/src/lib/libdrm/xf86drm.c: In function 'drmGetDevice2':
/.../contrib/libdrm-cacc86de9adf92f7dc7212928555d9e7f640da76/src/lib/libdrm/xf86drm.c:4758:33: error: assignment to 'drmPlatformBusInfoPtr' {aka 'struct _drmPlatformBusInfo *'} from incompatible pointer type 'char *' [-Wincompatible-pointer-types]
 4758 |        dev->businfo.platform    = addr;
      |                                 ^
```

ssumpf, 2025-05-06:
@cproc: [28f44b9](https://github.com/ssumpf/genode/commit/28f44b92cbdcc0d334b85f720ef0b331ff213105) fixes this (my issue5540_gcc14 branch)

cproc, 2025-05-06:
Thanks, marking the issue as fixed.

closed 2025-05-28 by chelmuth