genodelabs/genode #5551: mesa: compile errors with GCC 14

cproc, 2025-05-05:
When building `lib/mesa_gpu-softpipe` (or etnaviv, iris, lima) with GCC 14, the following errors occur:

```
/.../repos/libports/src/lib/mesa/softpipe/platform_softpipe.c:50:34: error: initialization of '_EGLSurface * (*)(_EGLDisplay *, _EGLConfig *, void *, const EGLint *)' {aka 'struct _egl_surface * (*)(struct _egl_display *, struct _egl_config *, void *, const int *)'} from incompatible pointer type 'void ** (*)(void **, void **, void *, const EGLint *)' {aka 'void ** (*)(void **, void **, void *, const int *)'} [-Wincompatible-pointer-types]
   50 |         .create_window_surface = dri2_genode_create_window_surface,
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/.../repos/libports/src/lib/mesa/softpipe/platform_softpipe.c:50:34: note: (near initialization for 'dri2_genode_display_vtbl.create_window_surface')
/.../repos/libports/src/lib/mesa/softpipe/platform_softpipe.c:51:34: error: initialization of '_EGLSurface * (*)(_EGLDisplay *, _EGLConfig *, void *, const EGLint *)' {aka 'struct _egl_surface * (*)(struct _egl_display *, struct _egl_config *, void *, const int *)'} from incompatible pointer type 'void ** (*)(void **, void **, void *, const EGLint *)' {aka 'void ** (*)(void **, void **, void *, const int *)'} [-Wincompatible-pointer-types]
   51 |         .create_pixmap_surface = dri2_genode_create_pixmap_surface,
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/.../repos/libports/src/lib/mesa/softpipe/platform_softpipe.c:51:34: note: (near initialization for 'dri2_genode_display_vtbl.create_pixmap_surface')
/.../repos/libports/src/lib/mesa/softpipe/platform_softpipe.c:53:28: error: initialization of 'EGLBoolean (*)(_EGLDisplay *, _EGLSurface *)' {aka 'unsigned int (*)(struct _egl_display *, struct _egl_surface *)'} from incompatible pointer type 'EGLBoolean (*)(void **, void **)' {aka 'unsigned int (*)(void **, void **)'} [-Wincompatible-pointer-types]
   53 |         .destroy_surface = dri2_genode_destroy_surface,
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/.../repos/libports/src/lib/mesa/softpipe/platform_softpipe.c:53:28: note: (near initialization for 'dri2_genode_display_vtbl.destroy_surface')
/.../repos/libports/src/lib/mesa/softpipe/platform_softpipe.c:55:26: error: initialization of 'EGLBoolean (*)(_EGLDisplay *, _EGLSurface *, EGLint)' {aka 'unsigned int (*)(struct _egl_display *, struct _egl_surface *, int)'} from incompatible pointer type 'EGLBoolean (*)(void **, void **, EGLint)' {aka 'unsigned int (*)(void **, void **, int)'} [-Wincompatible-pointer-types]
   55 |         .swap_interval = dri2_genode_swap_interval,
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~
/.../repos/libports/src/lib/mesa/softpipe/platform_softpipe.c:55:26: note: (near initialization for 'dri2_genode_display_vtbl.swap_interval')
```

ssumpf, 2025-05-06:
@cproc: [ec99e09](https://github.com/ssumpf/genode/commit/ec99e09cee4adab96db2beaf72b0676c837b5524) fixes all _lib/mesa_gpu-*_ targes for me (my issue5540_gcc14 branch).

cproc, 2025-05-06:
@ssumpf: I'm getting an error for `lib/mesa` now:

```
In file included from <command-line>:
/.../repos/libports/src/lib/mesa/include/platform.h:22:10: fatal error: egltypedefs.h: No such file or directory
   22 | #include <egltypedefs.h>
      |          ^~~~~~~~~~~~~~~
```

ssumpf, 2025-05-06:
@cproc: Ok, will check.

ssumpf, 2025-05-06:
@cproc: I'm getting missing  `type_traits` header. Caused by -nostdinc? Have you seen this before?

```
src/lib/mesa/src/util/enum_operators.h:27:10: fatal error: type_traits: No such file or directory
   27 | #include <type_traits>
      |          ^~~~~~~~~~~~~
```

cproc, 2025-05-06:
> [@cproc](https://github.com/cproc): I'm getting missing `type_traits` header. Caused by -nostdinc? Have you seen this before?

I just tried your branch in a different working directory and got this error too. Preparing the `stdcxx` port solved the problem. I also got an error about missing `xf86drm.h` because the corresponding `libdrm` version was not prepared.

ssumpf, 2025-05-06:
@cproc: Thanks!

ssumpf, 2025-05-06:
@cproc : [36fac10](https://github.com/ssumpf/genode/commit/36fac10e623ee87a5842c2aee7109ce08c0a6c86) compiles `lib/mesa` here.