genodelabs/genode #5559: Unify lx_emul Block bindings

cnuke, 2025-05-05:
The following commits are going to unify the various Block bindings found in the `block.c` file in the ported `sd_card` drivers.

nfeske, 2025-05-15:
I like this change, yet would appreciate a green light by @skalk before merging.

skalk, 2025-05-15:
I'm fine with the consolidation work. @cnuke very nice that you eliminated the need for the legacy linux port. Only question left: can't we delete that port as well?

skalk, 2025-05-15:
@cnuke thanks for the explanation. Well, I think it is more consistent and straight forward to now remove the legacy port and its patches, and later if needed to rename the `linux` port into `legacy_linux`, when adding a new one.

nfeske, 2025-05-16:
Merged to staging.

With legacy_linux removed, can #5264 be closed now? What about rpi asc mentioned there?

cnuke, 2025-05-16:
`rpi_linux` is still on version *6.1.37*.

ssumpf, 2025-05-16:
> `rpi_linux` is still on version _6.1.37_.

@cnuke: I have the RPI on my table and just put it on my TODO list :wink:

skalk, 2025-05-19:
Actually, if we want to maintain the upgrades of rpi_linux further, in my eyes it would be enough to upgrade it this time with the upcoming, road-map-scheduled upgrade of DDE Linux in the upcoming release, instead of upgrading now (6.1 ->  6.6) and then immediately again.

commit 2578bcac (2025-05-16): lx_emul: unify Block bindings

Consolidate the Genode-C-API Block users by having one implementation
in the lx_emul library.

Fixes #5559.