genodelabs/genode #5560: cxx: operator delete (...) called - not implemented

ssumpf, 2025-05-05:
With Sculpt 25.04 we experience two different scenarios where error messages like

`Error: cxx: operator delete (void *, unsigned long) called - not implemented`

appear.

1. on newer Ubuntu versions (e.g., 24.0.2 LTS) where _fb_sdl_ outputs many of these during startup
1. when deleting destructors are called for base-library classes (e.g., `Timer::Connection`)

ssumpf, 2025-05-05:
361dc5b addresses the first issue
1ff824b addresses the second issue

@chelmuth: What do you think?

chelmuth, 2025-05-06:
> [@chelmuth](https://github.com/chelmuth): What do you think?

I think that's great. Tested with run/demo and merged to staging instantly.

ssumpf, 2025-05-09:
[c686296](https://github.com/ssumpf/genode/commit/c6862961f2b23186e40a8c6d0efd174f01972727) fixes KERNEL=linux 32Bit

closed 2025-05-28 by chelmuth