genodelabs/genode #5561: Unify handling of buffer-exceeded error conditions

nfeske, 2025-05-06:
With the removal of C++ exceptions (#5245), we need a robust alternative for handling situations where generated data exceeds the size of a memory buffer. Notable places are `Xml_generator`, `Reporter`, `Session_requester`, and `Dynamic_rom_session`.

Since these utilities are related, they should share a common error type hosted at base/error.h:
```
enum class Buffer_error { EXCEEDED };
```
This type is meant to be used to form utility-specific result types. E.g., the result of the `Reporter`:
```
using Result = Attempt<Ok, Buffer_error>;
```
In contrast, the `Xml_generator` returns the number of generated bytes on success:
```
using Result = Attempt<size_t, Buffer_error>;
```
Thanks to the use of `Attempt`, which is marked as `[[nodiscard]]`, the compiler enforces that results are always evaluated. So no error condition remains unconsidered. Of course, one can deliberately decide to ignore errors by putting `(void)` in front of the function call that returns attempt. But these places become plainly visible now.

As result values cannot be returned form constructors, the direct construction and use of `Xml_generator` is no longer practical. Instead, a new class function `generate` provides the functionality as a plain function returning the result or error:
```
Xml_generator::Result const result = 
  Xml_generator::generate(byte_range_ptr, tag_name,
    [&] (Xml_generator &xml) {
      xml.node(...);
      xml.attribute(...);
      ... 
    });
```
The higher-level utilities like `Reporter` and `Dynamic_rom_session` should follow the same pattern.

By enforcing the error handling, many places come to attention that employed a `Reporter` without considering the buffer-exceeded case, in particular tiny state reports that are never expected to exceed 4 KiB. In most cases, the error handling can be avoided by using an `Expanding_reporter` instead. This way, the (unexpected) error may in the worst case end up in a resource request but the component won't ever abort due to an unhandled buffer-exceeded exception.

cproc, 2025-05-08:
Commit 55b3002 adapts Qt6.

cproc, 2025-05-08:
Commit c447a2f adapts Qt5.

nfeske, 2025-05-08:
Thanks a lot @cproc!

commit fe5d10f7 (2025-05-06): Unify handling of buffer-exceeded error conditions

- Introduce common Buffer_error type.
- Enforce the handling of buffer-exceeded errors.
- Unify interface of Reporter and Expanding_reporter,
  using Byte_range_ptr instead of basic types.
- Simplify the Xml_generator API by providing the plain
  class function Xml_generator::generate.
- Use Expanding_reporter where appropriate.

Fixes #5561
closed 2025-05-28 by chelmuth