genodelabs/genode #5568: Add the `log-session` to the black hole component

Washipp, 2025-05-10:
When all log-sessions are routed to the same place and displayed in a terminal, it can get cluttered when a lot of logging happens (for example when fuzzing). The black hole component currently does not provide the log session.

smhmeier, 2025-05-15:
Added the same feature in a new commit using a gapfruit account [7bce1a9](https://github.com/genodelabs/genode/commit/7bce1a92626d9b8837a3893e44775d2cbaf95fa3). This should solve the licensing issue, though a copyright notice could be required.

smhmeier, 2025-05-19:
@chelmuth see updated commit [892f0e6] (https://github.com/genodelabs/genode/commit/892f0e680040a5a9934f7f2e3ed20e4c76fb1786).

chelmuth, 2025-05-19:
Merged to staging.

fdb740f340c Add log-session to the black hole component
a4ca8275123 fixup "Add log-session to the black hole component" (Updated log.h according to chelmuth reply)
b8627c4e22b fixup "Add log-session to the black hole component" (adapt to staging, style, test)

closed 2025-05-28 by chelmuth