genodelabs/genode #5569: base-nova floods log with "ec_time failed" if there is a "zombie" component

jschlatow, 2025-05-12:
I accidentally messed up the ram attribute of my usb_webcam runtime. Having `ram="64"` instead of `ram="64M"`, the component was unsurprisingly unable to startup. However, I was not aware of my fault until I started top_view, which continuously triggered "ec_time failed" warnings from core as long as the wrongly attributed usb_webcam component was deployed.

I believe the best fix for this would be to only output the warning once (per thread). Are there other suggestions?

alex-ab, 2025-05-13:
Ideally, no syscalls on invalid capability selectors should be done at all. I added two commits, whereby the first tracks the fact that never an valid thread was created and therefore no CPU utilization data will be available. The second removes all error code checking for syscalls regarding ec/sc utilization times. Personally I prefer the first one, the second one may hide bugs. In principle also both are mergeable.

alex-ab, 2025-05-19:
@nfeske, @chelmuth: please consider b5ca88ec999dba16ca49ac7396ec78c8dd332134 for staging

nfeske, 2025-05-19:
Thanks @alex-ab for the reminder/hint. Merged to staging.

commit 92cfa160 (2025-05-19): nova: track successful creation of a thread

With this addition, we can avoid syscalls on invalid capabilities for EC and
SC utilization time if a thread construction failed.

Fixes #5569
closed 2025-05-28 by chelmuth