genodelabs/genode #5577: nitpicker: seq events are always forwarded to the most recently touched view owner

jschlatow, 2025-05-16:
In the scope of #5485, I already noticed that, after closing the popup dialog with a touch event, click events are ineffective in the Leitzentrale. Functionality can be restored by touching any button.

Looking into this issue in more detail, I noticed that the seq event for the non-functional clicks do not reach menu_view. They actually get stuck in nitpicker because it tries submitting the events to the most recently touched view owner:

https://github.com/genodelabs/genode/blob/a128bbd6b8dca7c7cc1d5583e9137626b1cc8d9d/repos/os/src/server/nitpicker/user_state.cc#L294-L300

In the code snippet above, `if (_touched)` always evalutes to true, however, `_touched->label()` is an empty string. It appears the `_touched` points to a bare `View_owner` object, not a `Session_component`.

@nfeske Do you have an explanation/solution? I have not found a proper place to set `_touched = nullptr`.

jschlatow, 2025-09-24:
@nfeske I'd like to fix this for the upcoming Sculpt release. Do you have any idea how to fix it?

nfeske, 2025-09-24:
Unfortunately, I've no more intuition than you. The mixed operation of touch and click is probably not carved out completely as I have never thought this through systematically. I'd appreciate you crawling into that rabbit hole and report what you find. ;-)

closed 2025-10-30 by chelmuth