genodelabs/genode #5579: qt6: compile errors with GCC 14

cproc, 2025-05-16:
When building `qt6/base` for x86_32 with GCC 14, errors like the following occur:

```
/.../contrib/qt6_base-e5ae92556716c9d5b41355cb845f196e1d378810/src/lib/qt6_base/src/corelib/global/qfloat16.h:64:52: error: SSE register return with SSE2 disabled
   64 |     constexpr operator NativeType() const noexcept { return nf; }
      |                                                    ^
/.../contrib/qt6_base-e5ae92556716c9d5b41355cb845f196e1d378810/src/lib/qt6_base/src/corelib/global/qfloat16.h: In function 'qfloat16 operator+(qfloat16, qfloat16)':
/.../contrib/qt6_base-e5ae92556716c9d5b41355cb845f196e1d378810/src/lib/qt6_base/src/corelib/global/qfloat16.h:124:147: error: operation not permitted on type '_Float16' without option '-msse2'
  124 |     friend inline qfloat16 operator+(qfloat16 a, qfloat16 b) noexcept { return qfloat16(static_cast<NearestFloat>(a) + static_cast<NearestFloat>(b)); }
      |                                                                                                                                                   ^
```

cproc, 2025-05-16:
Fixed by 4203cf0.

chelmuth, 2025-05-16:
Merged to staging.

commit 07d5ec19 (2025-05-16): qt6: fix compile errors with GCC 14

Fixes #5579