genodelabs/genode #5581: sel4: improve support

alex-ab, 2025-05-19:
Currently several static allocators in base-sel4 exists, which lead on very dynamic scenarios (like bomb and Sculpt) to unexpected results. Especially, if triggered in core it is fatal. Additonally, the manageable seL4 cspace is limited which leads to unexpected failures in core.

The issue is about improving the current state and if not possible, to document the limitations.

alex-ab, 2025-05-19:
@nfeske: the commits on issue_5581 improves the current state already, at least run/bomb on hardware succeeds for me now. Still, a lot of improvements are required.

nfeske, 2025-05-19:
Thanks for this very fine commit series, which I happily merged to staging.

alex-ab, 2025-05-22:
@nfeske: 4c0e4a24870dc4c139d5a4a979f04c38fa1f56d6 and 3ae59f5f2d1fc0bd139ee1e98b1fdcbb203a4d58 please consider for staging

nfeske, 2025-05-22:
@alex-ab wow! That's an impressive re-work. I merged both commits to staging.

As a minor coding-style detail, we usually don't leave a space between `&` and the variable name. The reason is that the `&` belongs more closely to the variable than to the type (I know, that's couter-intuitive, but that's the language).

alex-ab, 2025-05-23:
> As a minor coding-style detail, we usually don't leave a space between `&` and the variable name. The reason is that the `&` belongs more closely to the variable than to the type (I know, that's couter-intuitive, but that's the language).

Okay, i added two fixup commits.

nfeske, 2025-05-23:
Much much appreciated. :-) Merged to staging.

alex-ab, 2025-06-06:
@nfeske, @chelmuth: can you please check/consider my issue_5581 branch for staging

nfeske, 2025-06-17:
Thanks @alex-ab for the welcome improvements. Good to see the assertions go. Merged to staging.

alex-ab, 2025-08-19:
@nfeske: please consider the two top most commits for staging. If the seL4 update to 13 causes too much trouble in the CI, we may also just revert it

nfeske, 2025-08-19:
Maybe, we can consider meeting somewhere in the middle, adjusting the quota of the installable pkgs (system shell etc.) but keeping the values of static sculpt system untouched?
This way, the building of an seL4 sculpt-image requires a patch, but the installable depot pkgs published by genodelabs would become compatible with seL4.

closed 2025-08-28 by nfeske