genodelabs/genode #5582: window layouter: close button is ineffective on maximized windows

jschlatow, 2025-05-19:
Today, I noticed that the window decorator's close button is ineffective when a window is maximized. It appears that it never worked or that it has been broken for quite some time. I've went through all the Sculpt versions from the genodelabs depot back to 23.10: They all show the same behaviour.

I briefly went through the layouter's code. If I understand it correctly, the closing is done at the end of a dragging action when the closer element was dragged. In other words: When the mouse button is released and the hovered element at the time of the press event was the closer window element. However, I could not locate why it is handled differently when the window is maximized.

@nfeske If you have any insight, let me know. Otherwise, I can dig into this when time permits.

jschlatow, 2025-05-19:
> The closing of a window is expressed as a resize of the window to size 0x0. I guess that this "resizing" contradicts the notion of "maximized", which somewhat pins the window size to the available target area. We would need to give the closed "size" precedence over the maximized size.

Thanks, that makes sense and was actually very helpful. Fixed by 1bcca33.

jschlatow, 2025-05-19:
The corresponding pkg archive is available on my depot (jschlatow/pkg/window_layouter/2025-05-19).

nfeske, 2025-05-20:
Thank you @jschlatow. Merged to staging (with a very minor style change).

commit f88e1039 (2025-05-20): window_layouter: make maximized windows closable

Closing a window is reflected by a resize to 0x0. As the 'maximized' state
of a window prevented any resize, closing was impossible.

Fixes #5582