genodelabs/genode #5583: sculpt: download_debug_info does not start

cproc, 2025-05-19:
When I start the Sculpt Options `download_debug_info` launcher, the `Diagnostics` window prints "download debug info requires" and the component does not start.

When I replace `<child identity="default_fs_rw"/>` by `<child name="default_fs_rw"/>` in `repos/gems/sculpt/launcher/download_debug_info` the component starts, but I' m not sure if this is the correct fix.

nfeske, 2025-05-19:
Thanks for bringing the issue to light. This is the correct fix. The `identity` attribute merely rewrites the client's label. For the routing decision, the child name is important. I guess that the lack the name attribute has the effect that the client depends on a (non-existing) child named "".

cproc, 2025-05-19:
Okay, fixed by 39a52fd.

commit d5f0d89c (2025-05-20): download_debug_info: fix "used ->" route

Fixes #5583
closed 2025-05-28 by chelmuth