genodelabs/genode #5587: libc: GCC-14 issue

ssumpf, 2025-05-20:
Currently some tests (e.g., _run/vfs_import_) on  x86_32, arm_v7, and arm_v8 (not x86_64) cause page faults in libc. I investigated the x86_32 issue and found that https://github.com/genodelabs/genode/blob/d887fcc6291163db23e982dfd7bd3b76c0dc76cc/repos/libports/src/lib/libc/internal/kernel.h#L441 was removed (or optimized out) from the libc.lib.so binary by the 25.05 tool chain (verified with gdb and disassembly). When compiling libc with the 23.05 tool-chain the line is present. Since `_state` is not updated this breaks our libc-execution model.

@cproc: Do you have any idea what is going on might this be related to `call_func` being `[[noreturn]]` and `__builtin_unreachable()`?

Related to #5540

ssumpf, 2025-05-20:
Removing `[[noreturn]]` and `__builtin_unreachable()` from `call_func` seems to leave the `_state` line intact.

ssumpf, 2025-05-20:
Leaving `[[noreturn]]` intact and replacing `__builtin_unreachable()` by `Genode::sleep_forever` or `while (true)` also works.

ssumpf, 2025-05-21:
> It seems visibly using `__builtin_unreachable()` in the implementation of `call_func()` misleads the compilation optimizer to draw wrong conclusions. How about moving the asm from the headers into a compilation unit?

@chelmuth: [72b47a7](https://github.com/ssumpf/genode/commit/72b47a746e0eb2a4fa9e6dcc4004929f726f9e8d) moves the `call_func` implementation from the headers into compilation units. It works on all supported platforms so far.

chelmuth, 2025-05-22:
Sounds good, may we remove `INC_DIR += $(REP_DIR)/src/lib/libc/spec/arm` from all libc.mk files then?

chelmuth, 2025-05-22:
You're right, they are all gone. Don't know what I saw this morning. Merged to staging.

nfeske, 2025-05-23:
Strange, the CI reports that vfs_import.run still stumbles on 32-bit.

ssumpf, 2025-05-26:
> Strange, the CI reports that vfs_import.run still stumbles on 32-bit.

@chelmuth: Do you have a log or more information for me on this?

commit 7f98e94f (2025-05-22): libc: move call_func implementation from header

When implemented as inline header function, __builtin_unreachable seems
to trigger dangerous compiler optimizations, like removing the '_state =
USER' because the compiler assumes a non-returning branch.

fixes #5587