genodelabs/genode #5589: Unify fixed-width C types

cproc, 2025-05-20:
Currently, some fixed-width C types like `int64_t` differ between 32-bit and 64-bit architectures with regard to `long long` vs. `long` or between GCC, libc and Genode base, which could cause compile errors or ABI incompatibilities.

Related to #5431.

cproc, 2025-05-20:
Fixed by f0683ca.

The tool chain needs to be rebuilt for the GCC changes.

With this commit, the actual types are defined in the `genode-stdint.h` GCC header and the other places use the corresponding built-in GCC type macros.

Also, it looks like `repos/base/include/spec/32bit/base/fixed_stdint.h` and `repos/base/include/spec/64bit/base/fixed_stdint.h` could be merged since apparently only the comments differ?

nfeske, 2025-05-21:
Merged to staging.

commit 48d8e1e9 (2025-05-21): Unify fixed-width C types

Fixes #5589