genodelabs/genode #5593: extract: let final dir appear only after success

nfeske, 2025-05-22:
When the extract tool fails midway with the extraction of archive, the top-level directory of the incompletely extracted content is left over. From the outside it looks like the extraction is done. In particular, the download_manager  of Sculpt OS wrongly interprets the situation where an archive extraction failed as a completed installation on the archive. By extracting the content to a differently named directory (e.g., suffixed with ".incomplete") and renaming the directory to its designated name as a final step, such misinterpretations could be prevented.

nfeske, 2025-09-25:
The commit series on my [failsafe_extract](https://github.com/nfeske/genode/commits/failsafe_extract/) branch safeguards the extraction of depot archives. The commit message of https://github.com/nfeske/genode/commit/12cc2858492027697a713cad61e7c133af4a4722 goes into detail about the three classes of failures addressed.

sidhussmann, 2025-10-02:
Thank you @nfeske for the concise implementation and the careful crafted commits. This helped me backport the commits to our 25.05-based branch. Testing with `pkg` archives, the extract/commit mechanism works like a charm. However, with `image` archives, the `commit` step prints a warning:

```
[init -> static -> log -> os -> boot -> depot_download -> dynamic -> verify] /public/gapstage/image/gapfruit_os-imx8mp_iot_gate-2025-09-12.tar.xz: good signature
[init -> static -> log -> os -> boot -> depot_download -> dynamic] child "stage" exited with exit value 0
[init -> static -> log -> os -> boot -> depot_download -> dynamic -> extract] extracted '/public/gapstage/image/gapfruit_os-imx8mp_iot_gate-2025-09-12.tar.xz' to '/depot/gapstage/image/extract'
[init -> static -> log -> os -> boot -> depot_download -> dynamic] child "extract" exited with exit value 0
[init -> static -> log -> os -> boot -> depot_download -> dynamic -> commit] Warning: skip removal of non-empty dir gapstage/image/extract
[init -> static -> log -> os -> boot -> depot_download -> dynamic] child "commit" exited with exit value 0
[init -> static -> log -> os -> boot -> depot_download -> manager] installation complete.
```

When investigating the file system content, it turns out that the following path exists:

```
depot/gapstage/image/extract/gapfruit_os-imx8mp_iot_gate-2025-09-12/image.itb
```

Shouldn't `fs_tool` remove the `extract` directory recursively?

sidhussmann, 2025-10-02:
> I deliberately decided against wiping the extract directory to aid the investigation of anomalies (like this one?).

Sounds reasonable.

> Could you check that the final image/gapfruit_os-<version>/ directory actually exists and its content is complete?

I can confirm that the image exists and is complete:

```
noux@/rw/boot/depot/gapstage/image/gapfruit_os-imx8mp_iot_gate-2025-09-12> ls -hal                                                                                                                                                                                          
total 122K
-rwx------ 1 root 0 16M Jan  1  1970 image.itb
```

sidhussmann, 2025-10-03:
I can confirm that it also happens with ram fs. It looks as if the problem arises because `fetchurl` downloads the same image archive twice. It reminds me of #4999. 

Attached log will reveal 3 cases:
1. `fetchurl` is started twice and the first time `commit` doesn't print a warning. Although other bad stuff happens that is out of scope for this discussion.
2. `depot_download` first takes the downloaded archive from `public`, successfully extracts and commits it, but then re-triggers the download/verify/extract/commit process, which leads to the log message `commit] Warning: skip removal of non-empty dir gapstage/image/extract`
3. Similar as in step 2, although `public` is empty and the image archive is downloaded twice.

[small.log](https://github.com/user-attachments/files/22676018/small.log)

nfeske, 2025-10-03:
Thanks a lot @sidhussmann for the valuable insights and for drawing the connection to #4999. This looks like a perfect opportunity to address #4999. ;-) I'll get to it once I'm back on deck. In the meantime, please bear with the warning message.

sidhussmann, 2025-10-22:
@nfeske our CI revealed that `depot_download.run` fails. And indeed, it also fails on `genodelabs/staging` with HEAD= 83b3a36dd70a1bf3fb434030aee42c1f274442fc:

```
$ make -C build/x86_64/ KERNEL=hw BOARD=pc run/depot_download
...

[init -> report_rom] report 'depot_download -> state'
[init -> report_rom]   <state progress="yes">
[init -> report_rom]    <archive path="genodelabs/api/base/2019-03-17" state="stage"/>
[init -> report_rom]    <archive path="genodelabs/api/os/2019-03-17" state="stage"/>
[init -> report_rom]    <archive path="genodelabs/api/gems/2019-03-17" state="stage"/>
[init -> report_rom]    <archive path="genodelabs/api/input_session/2019-02-25" state="stage"/>
[init -> report_rom]    <archive path="genodelabs/api/nitpicker_session/2019-02-25" state="stage"/>
[init -> report_rom]    <archive path="genodelabs/api/framebuffer_session/2019-02-25" state="stage"/>
[init -> report_rom]    <archive path="genodelabs/api/report_session/2019-02-25" state="stage"/>
[init -> report_rom]    <archive path="genodelabs/api/timer_session/2019-02-25" state="stage"/>
[init -> report_rom]   </state>
[init -> depot_download -> manager] Error: Uncaught exception of type 'Depot::Archive::Unknown_archive_type'
[init -> depot_download -> manager] Warning: abort called - thread: ep
Error: Test execution timed out while waiting for ".*path="genodelabs/pkg/wm/2019-03-17" state="done".*.*path="genodelabs/bin/x86_64/backdrop/2019-03-17" state="done".*.*path="genodelabs/index/19.02" state="done".*.*path="genodelabs/index/19.03" state="failed".*"
make: *** [Makefile:446: run/depot_download] Error 254
make: Leaving directory '/home/user/misc-code/genode-upstream/build/x86_64'
```

nfeske, 2025-10-22:
I just gave the depot_download.run script a try on Qemu/PC via NOVA and base-hw. It looks happy to me.

sidhussmann, 2025-10-23:
@nfeske I keep two genode repositories, so I can switch between upstream and our branch without having to rebuild everything. I must have messed up with the `create_build_dir` step, because the `Makefile` symlink within the upstream build directory somehow was pointing to the other build directory. I can confirm: `depot_download.run` works on staging. Regarding our CI, I made a mistake in backporting to our 25.05-based branch. I'm sorry for the confusion. Thank you for your efforts. I consider this issue resolved.

ssumpf, 2025-10-23:
> [@nfeske](https://github.com/nfeske) I keep two genode repositories, so I can switch between upstream and our branch without having to rebuild everything. I must have messed up with the [...]

@sidhussmann: Since this is the 2nd time something like this happened, I would very much appreciate if you could double check everything before reporting, meaning, make sure you have a clean build, make sure you don't have outdated packets, and make sure there are no changes (or only known changes) to the repository applied, because premature reporting means additional work for us. This would really help. Thank you for your understanding.

commit beb314af (2025-09-29): Improve fail-safety of depot-download extract step

- Continue with the next job after encountering a failed extraction
  step. Previously, the installation stopped with an error message, but
  left potentially remaining installation items uninstalled.

- Prevent installation of incompletely extracted archives. The archive
  extraction can fail for several reasons like a malformed archive
  format, or a full disk, or a bug in the extract tool. In such cases,
  the extracted content may be incomplete. With this patch, archives are
  extracted at a staging area at <depot>/<user>/<type>/<name>/ex [... cut]