genodelabs/genode #5605: sculpt: differentiate priority of runtime init and its children

skalk, 2025-06-13:
Nowadays, Sculpt OS provides two priority levels at the top-level init, and four priority levels at the second-level runtime init. The latter divide runtime components into: driver, multimedia, default, and background. To be able to distinguish in between top-level lower priority components (e.g. the runtime-init) and the four different levels of runtime components, we actually need to provide more than 4-levels within the runtime-init. Thereby the highest priority is reserved for the runtime-init itself, and driver, multimedia etc. get priorities: -1, -2, ...

This issue came up when trying to solve issue #5117 and trying to map runtime-init to the multimedia-group, but runtime components of level 'driver' to the driver group. Until now, both are indistinguishable.

nfeske, 2025-06-19:
Merged https://github.com/skalk/genode/commit/dacdb3f3b2710213d560aa02d2ff5d4e0db040d0 to staging.

closed 2025-08-28 by chelmuth

chelmuth, 2025-10-24:
@skalk *run/scale_priority* fails since 2025-10-21 with the following output. May commit d2951cba3181de72f4ed0b6e015c24c9e9b21e4e be accountable?

```
[2025-10-21 09:25:50] [init -> test-scale_priority] [31mError: Unexpected priority of component second lowest expected 2 got 1
[2025-10-21 09:25:50] [init -> test-scale_priority] [31mError: Unexpected priority of component second highest expected 1 got 0
```

skalk, 2025-10-24:
Yes, that is right. Scale priority test that assuming linear increasing or decreasing "priorites" does not align anymore with the "fixed Sculpt priorities to scheduling group ids" anymore. I think we should disable it in the test.