genodelabs/genode #5607: riscv: ldso fails to resolve weak undefined symbols in stdcxx

ssumpf, 2025-06-17:
When executing

`make KERNEL=hw BOARD=virt_qemu_riscv TEST_PKGS=test-stdcxx TEST_SRCS= run/depot_autopilot`

we experience errors like

`Error: LD: symbol not found: 'transaction clone for operator new[](unsigned long)'`

This issue seems caused by weak undefined symbols of _libitm_ and seems to be related to https://gcc.gnu.org/bugzilla/show_bug.cgi?id=119369. On x86 these symbols will end as undefined functions causing R_JUMP_SLOT relocation. Because the functions are not called, there is no problem. On RISC-V these symbols have no type, they are not functions and  R_RISCV_64 relocations are emitted by binutils-ld. These relocations are used for global data and resolved during library load, by searching the loaded library only. Because the symbol cannot be resolved the test fails.

ssumpf, 2025-06-17:
3996106 addresses the problem in ldso by ignoring weak undefined symbols for R_RISCV_64 relocations only. This issue should be kept open until a proper fix can be found upstream, either in gcc or binutils.

chelmuth, 2025-06-17:
I merged the commit, but what I actually want to achieve is the following.

```diff
+++ b/repos/libports/include/stdcxx/bits/os_defines.h
@@ -32,4 +32,7 @@
  */
 #define _GTHREAD_USE_MUTEX_INIT_FUNC 1
 
+/* no support for referencing weak symbols without a definition */
+#define _GLIBCXX_USE_WEAK_REF 0
+
 #endif /* _INCLUDE__STDCXX__BITS__OS_DEFINES_H_ */
```

@cproc is this file still used?

commit 57fca430 (2025-06-17): stdcxx: prevent refs to weak symbols w/o definition

When executing

  make KERNEL=hw BOARD=virt_qemu_riscv TEST_PKGS=test-stdcxx TEST_SRCS=run/depot_autopilot

we experienced errors like

  Error: LD: symbol not found: 'transaction clone for operator new[](unsigned long)'

This issue seems to be caused by weak undefined symbols of libitm and
seems to be related to

  https://gcc.gnu.org/bugzilla/show_bug.cgi?format=multiple&id=69506

Thus, we define _GLIBCXX_USE_WEAK_REF 0 like its done in newlib and
darwin environments and, thereby, drop ITM support.

Fixes #5607
closed 2025-08-28 by chelmuth