genodelabs/genode #5609: acpica: Error: illegal READ

rite, 2025-06-20:
At least with QEMU, `run/acpica` fails with the following log output (here with base-hw, but it also fails with base-nova):
```
Genode 25.05-1-g952ab94352
755 MiB RAM and 63256 caps assigned to init
[init -> acpica]   0x1000000 .. 0x14ffffff: linker area
[init -> acpica]   0x40000000 .. 0x4fffffff: stack area
[init -> acpica]   0x30000 .. 0x15cfff: ld.lib.so
[init -> acpi] Found MADT
[init -> acpi] MADT IRQ 0 -> GSI 2 flags: 0
[init -> acpi] MADT IRQ 5 -> GSI 5 flags: 13
[init -> acpi] MADT IRQ 9 -> GSI 9 flags: 13
[init -> acpi] MADT IRQ 10 -> GSI 10 flags: 13
[init -> acpi] MADT IRQ 11 -> GSI 11 flags: 13
[init -> acpi] Found MCFG
[init -> acpi] MCFG BASE 0xb0000000 seg 0x0 bus 0x0-0xff
[init -> acpi] RSDT OEM 'BOCHS ', table id 'BXPC    ', revision 1, creator 'BXPC' (1)
[init -> acpi] SMBIOS table (entry point: 0x166560 structures: 0xf6580)
[init -> event_dump] Input event #0     PRESS KEY_BATTERY 65534 key count: 1
[init -> event_dump] Input event #1     RELEASE KEY_BATTERY     key count: 0
[init -> ps2] Using keyboard with scan code set 1 (xlate)
[init -> acpica -> debug] ACPI: RSDP 0x0000000000000001 000024 (v2 Faked )
[init -> acpica -> debug] ACPI: RSDT 0x0000000031fe2351 000038 (v1 BOCHS  BXPC     00000001 BXPC 00000001)
[init -> ps2] detected ExPS/2 mouse - activating scroll-wheel and 5-button support
[init -> acpica -> debug] ACPI: FACP 0x0000000031fe2149 0000f4 (v3 BOCHS  BXPC     00000001 BXPC 00000001)
[init -> acpica -> debug] ACPI: DSDT 0x0000000031fe0040 002109 (v1 BOCHS  BXPC     00000001 BXPC 00000001)
[init -> acpica -> debug] ACPI: FACS 0x0000000031fe0000 000040
Error: illegal READ at address 0x17379 by pager_object: pd='init -> acpica' thread='ep' ip=0x105f180
```

`git bisect` shows that this is a regression introduced with 1ef80e86e29292538bdc5a2401b9f3cb7d42562a.

rite, 2025-06-20:
605c0a8 fixes the missing dynamic linker in the run scenario. Regarding the illegal READ error, I have no clue 😳 ...

alex-ab, 2025-06-26:
@rite: thanks for the bisecting, with that I had a clear idea where to look to. I added a fix and added a commit to add the test to the autopilot

rite, 2025-06-27:
Perfect, thank you @alex-ab!

commit 54568f33 (2025-06-27): acpica: avoid auto deallocation in attach()

Fixes #5609