genodelabs/genode #5675: chroot: fails to create File_system session for certain label/paths

sidhussmann, 2025-08-30:
I encountered a weird behavior in the `File_system` session establishment between a `chroot` component and a `vfs` server. I created a run script that reproduces the error in a simplified scenario on base-linux in TODO_COMMIT_ID.

The run script succeeds with the following routes of the `test` component:

```xml
			<service name="File_system" label_prefix="home ->"> <child name="chroot_test_fs" identity="os -> dynamic -> someapp" resource="/conf"/> </service>
			<service name="File_system" label_prefix="home ->"> <child name="chroot_test_fs" identity="os -> dynamic -> someapp" resource="/configt"/> </service>
			<service name="File_system" label_prefix="home ->"> <child name="chroot_test_fs" identity="os -> dynamic" resource="/someapp/config"/> </service>
```

However, it fails with the following routes:

```xml
			<service name="File_system" label_prefix="home ->"> <child name="chroot_test_fs" identity="os -> dynamic -> someapp" resource="/confi"/> </service>
			<service name="File_system" label_prefix="home ->"> <child name="chroot_test_fs" identity="os -> dynamic -> someapp" resource="/config"/> </service>
```

You can verify this by uncommenting the respective lines in the run script. 

As you can observe, the error is not merely a result of overstepping the session argument limits, as the issue https://github.com/genodelabs/genode/issues/5527 addressed: The resulting good label lengths are both smaller and larger than the two bad labels. 

What confuses me is that the `chroot` component seems to assemble the session arguments correctly (see the debug outputs via the subsequent commits on my branch). However, when the session reaches the `vfs` server, the label turns from `rw -> /subdir/os/dynamic/someapp/conf/` (good case) to `rw -> 0` (bad case). It almost looks like there is a bug in the parsing of the session arguments.

nfeske, 2025-09-01:
Without having looked at this issue description in detail yet, the `resource` attribute values of the posted examples caught my eye. To denote a directory, the values must start *and end* with a slash. This is briefly mentioned in the [documenation](https://genode.org/documentation/genode-foundations/25.05/components/Common_session_interfaces.html#File_system).

sidhussmann, 2025-09-01:
> Without having looked at this issue description in detail yet, the `resource` attribute values of the posted examples caught my eye. To denote a directory, the values must start _and end_ with a [...]

Thanks @nfeske! That's good to know. I adapted the routes and verified that it doesn't change the behavior. `chroot` makes sure that a `/` is appended during session establishment.

nfeske, 2025-09-17:
Given your descritpion, I suspected that your scenario hits the limit of the session args. The session args created by the chroot component are around 143 bytes ("tx_buf_size=131072, cap_quota=12, ram_quota=195789, diag=0, writeable=1, label="os -> dynamic -> someapp -> /subdir/os/dynamic/someapp/config/"). When the session request arrives at init, the label is further prefixed by the identity of the chroot component "chroot_test_fs -> ", adding 19 bytes. The result hardly or no longer fits into the `Session_args` type. The issue disappears when shortening "dynamic" as "d". It also disappears when shortening "chroot_test_fs" as "c".

But I wondered where does the strange `0` comes from? It turns out that the call of `Arg_string::set_arg` at https://github.com/genodelabs/genode/blob/master/repos/base/src/lib/base/child.cc#L156 overwrites the first byte of the `label` variable with the zero-termination of the argbuf string, which is due to an off-by-one in the sanity check of `Arg_string::add_arg`. With the fix of commit 92c07040b05af97c557f12301bc1865cea309544, your run script succeeds.

Thank you @sidhussmann for uncovering and reporting this bug!