genodelabs/genode #5689: depot_deploy: pkg/runtime vs deploy config precidence

sidhussmann, 2025-09-18:
A config node defined within a runtime file of a pkg archive takes precedence over what is defined in the deploy file.

Currently, the priorities of the config for a start node within the deploy mechanism are as follows:

1. The `config` sub-node of a start node within the deploy file
2. The `config` sub-node of the respective `pkg/runtime` file
3. The `ROM` route with `label="config"`

Is this intentional? As a user, I would intuitively expect the following priorities:

1. The `config` sub-node of a start node within the deploy file
2. The `ROM` route with `label="config"`
3. The `config` sub-node of the respective `pkg/runtime` file

The reason is twofold: As a user it is not obvious that the `pkg/runtime` contains a `config` node, nor what it actually contains since it is quite hidden. From a security perspective, I want to give more trust to the `deploy` file than to whatever the pkg maintainer/depot user thought reasonable for the pkg runtime.

I noticed this behavior by utilizing the `trace_recorder` pkg, which contains an example config within the pkg runtime. Adding the following start node to the deploy file results in the example config being loaded:

```xml

	<start name="trace_recorder" pkg="insidioussid/pkg/trace_recorder/2025-09-18T1339">
		<heartbeat/>
		<route>
			<service name="File_system" label="trace -> /">
				<parent identity="trace_recorder"/>
			</service>
			<service name="TRACE">
				<parent label="os"/>
			</service>
			<service name="ROM" label="config">
				<parent label="config -> os/trace_recorder"/>
			</service>
			<service name="ROM" label="platform_info">
				<parent/>
			</service>
			<service name="RTC">
				<parent/>
			</service>
		</route>
	</start>

```

If I remove the `config` node from the `pkg/trace_recorder/runtime`, the above example works as intended.

sidhussmann, 2025-09-19:
You are right. Reading the first sentence again is indeed confusing. I tried to summarize the requirement that I want to generally trust the deploy file more than the pkg/runtime.

> A `<config>` defined at the deploy file always wins.

This is correct. 

> Your question seems to come down to the priority of a ROM route defined at /config/deploy versus the `<config>` found in the pkg/runtime.

This nails it, yes. 

> Rewiring the config ROM route is quite rare in Sculpt but an example is the nano3d preset, which routes the config of the nano3d pkg to a dynamic-ROM component. But this scenario seems to match your intuition, doesn't it?

The `nano3d` pkg archive is a good example. Its runtime config does not contain a `<config>` node, and therefore it behaves as I would expect. We want to separate the configuration of the component from what actually launches the component. And it turns out, currently our workaround, e.g., with the `trace_recorder` pkg archive, is to copy the pkg recipe to a new name and remove the `<config>` node within the `runtime` file, which is unfortunate.

> What I wrote above applies for Sculpt 25.04. Have you looked at a more recent version? If yes could this be a regression?

I assume you mean a regression regarding "A `<config>` defined at the deploy file always wins."? Then no, this is also true on our 25.05-based branch. 

I wonder if rewiring the config ROM route should be prioritized more than the `<config>` node within the `runtime` file.

nfeske, 2025-09-22:
> The nano3d pkg archive is a good example. Its runtime config does not contain a `<config>` node, and therefore it behaves as I would expect.

But the nano3d runtime does in fact *have* a `<config>` node, see: https://github.com/genodelabs/genode/blob/master/repos/gems/recipes/pkg/nano3d/runtime#L5

nfeske, 2025-09-22:
Just noticed that there also exists an _unconfigured_nano3d_ pkg: https://github.com/genodelabs/genode/blob/master/repos/gems/recipes/pkg/unconfigured_nano3d/runtime, which is used by the nano3d preset. Sorry for the confusion.