genodelabs/genode #5691: gcc: support -shared link flag on arm_v8a/riscv

ssumpf, 2025-09-22:
The `-shared` link option currently produces executables instead of shared libraries on arm_v8a and riscv targets. As suggested by @cproc changing `%(shared:-shared)` to `%{shared:-shared}` in `LINK_SPEC` fixes this for both targets. Surprisingly the `(` version works on arm_v7a (beats me).

ssumpf, 2025-09-22:
[9d569a3](https://github.com/ssumpf/genode/commit/9d569a3cab3f3f69277a8e49732c1698a2c58367) fixes the issue. I have tested this by modifying the linking of shared libraries to use `gcc` on Genode. Note: This is meant to be used in future toolchain updates since the Genode-build system does not rely on the `-shared` gcc option.

cproc, 2025-09-24:
> I merged the commit to staging. [@cproc](https://github.com/cproc) do you agree?

yes